hyperspy/rosettasciio #5: Logo for RosettaSciIO

jlaehne, 2022-07-25:
Was just playing with an idea for a logo for RosettaSciIO, based on some hieroglyphs (see https://de.wikipedia.org/wiki/%C3%84gyptische_Hieroglyphen). Symbols for R S TT I are used:

[image]

Dark and bright versions, with either grayshades in the text or pure b/w.

jat255, 2022-07-26:
Wikimedia has a nice 3D model of the real rosetta stone: https://commons.wikimedia.org/wiki/File:Rosetta_Stone.stl Perhaps we could work that into a logo idea somehow? It would get across the idea of "translation"

jlaehne, 2022-07-26:
Yes, thought about at least playing with the contour of the stone, which I find clearer than the 3D object (a bit complicated for a logo):

[image]

The 3D object would be under [CC-BY-NC-SA](https://www.myminifactory.com/object-licensing), while the contour was derived from a public domain work: https://commons.wikimedia.org/wiki/File:Rosetta_Stone_BW.jpeg

ericpre, 2022-08-16:
Thanks @jlaehne, For a square logo, personally, I find line 4, column 1 and 3 in https://github.com/hyperspy/rosettasciio/issues/5#issuecomment-1195326090 quite nice. :) 

It is possible to make one that work on both dark and light background?

jat255, 2022-08-16:
My concern is that the "hieroglyphs" don't make it obvious what this library does or is even called. I like the concept of using the stone, but maybe we make the name more obvious? 

Some quick mockups:

(I actually quite like this)
[image]

(I don't love this next one... but as more of a logo idea)
[image]

(maybe we make the acronym 'RSIO'?)
[image]

Here's an SVG with those mockups in case anyone else wants to play:

[image]

CSSFrancis, 2022-08-16:
@jat255 Those look pretty good.  I think the top one is the best as far as maybe something like a banner.  As far as a logo goes I think that we could probably just have the Rossetta stone with IO on it.

Maybe we think about adding some science "hieroglyphs" so that all parts of the name are part of the logo, something like a Bohr model of atom, a bandgap or a beaker etc.

jlaehne, 2022-09-05:
Lets narrow down the choices for the Logo, please everyone vote using thumbs up/down on the following questions:

jlaehne, 2022-09-05:
Black/white and white/black (thumbs up)
versus background-agnostic gray scale (thumbs down)

jlaehne, 2022-09-05:
greek letters instead of a/e (thumbs up)
versus stick with latin letters (thumbs down)

jlaehne, 2022-09-05:
Font from png in https://github.com/hyperspy/rosettasciio/issues/5#issuecomment-1217620339 - jlaehne (thumbs up)
versus font from png in https://github.com/hyperspy/rosettasciio/issues/5#issuecomment-1217054931 - jat255 (thumbs down)

ericpre, 2022-09-07:
@jordiferrero, interested to do some of your magic? :)

jordiferrero, 2022-09-08:
Thanks @ericpre for flagging this :-) I really love the suggestions already proposed. In fact, I was envisioning the exact same idea... the stone with the "IO" as writing.
I personally love simple flat design. So I took @jat255 [simple design](https://github.com/hyperspy/rosettasciio/issues/5#issuecomment-1217054931) and simply added the "Python" snake in the S of RosettaSci... as a hint of it being a Python package. Attached the .svg files in case you want to add it.

_Note: I did not know what font you used, so I vectorised the font, that's why it's not prefect..._

[image]
[image]

jordiferrero, 2022-09-08:
> Thanks @ericpre for flagging this :-) I really love the suggestions already proposed. In fact, I was envisioning the exact same idea... the stone with the "IO" as writing. I personally love simple [...]

[image]

jat255, 2022-09-08:
> I did not know what font you used

It's Lato: https://www.latofonts.com/

ericpre, 2022-09-10:
Thank you @jordiferrero. Time for a poll of two questions then!

ericpre, 2022-09-10:
normal S (thumbs up)
S with snake (thumbs down)

ericpre, 2022-09-10:
IO only inside the stone (thumbs up)
more letters in the stone (RSIO, RScIO) (thumbs down)

hakonanes, 2022-09-10:
don't know if it's a problem really, but might there be people reading "IO" as "ten"?

ericpre, 2022-09-10:
Yes good point, it may be worth considering `I/O` or lower case `io`. Can someone have a go at making different versions, please? :)

jlaehne, 2022-09-11:
> Yes good point, it may be worth considering `I/O` or lower case `io`. Can someone have a go at making different versions, please? :)

There should be fonts, like the one of these comments, which are Sans Serif, but still have short horizontal lines on the upper case I, which would make it clear to be IO and not 10.

jlaehne, 2022-09-12:
Though, I do like Lento, @jordiferrero would it be possible to add the horizontal lines on the I (like in Tahoma, Verdana, Noto, Segoe UI) on your design otherwise keeping the font?

jordiferrero, 2022-09-12:
> Though, I do like Lento, @jordiferrero would it be possible to add the horizontal lines on the I (like in Tahoma, Verdana, Noto, Segoe UI) on your design otherwise keeping the font?

[image]

Some iterations on your ideas

francisco-dlp, 2022-09-12:
+1 for the original (f)—I just find its minimalism more appealing.

Would it be possible (and good) to keep the space between characters uniform to emphasize that it is a single word? Currently IO seems separated from the rest.

francisco-dlp, 2022-09-12:
Talking about minimalism, what about remove "osetta" from the logo i.e. RSciIO with IO inside the stone?

jat255, 2022-09-12:
> what about remove "osetta" from the logo

As a logo for use in presentations, documentation, etc., I'd rather have the full name in the logo. I was partial to a smaller "RS/IO" in the stone as a "square" format logo (I also think the import should be `rsio`, but that ship has probably sailed :smile: ), but I understand if that's not a shared opinion.

jlaehne, 2022-09-12:
> (I also think the import should be `rsio`, but that ship has probably sailed)

v0.1 has not been released yet, so I would not say that it has sailed, but we should nail it now.

The double 'i' in the import is prone to typos, so that would be an argument for the even shorter version.

ericpre, 2022-09-12:
I agree that it is worth considering `rsciio` versus `rsio`, as it is typo prone.

When asking around about `IO` versus `io`, people who are not familiar with what `IO` means read it as 10... so even if option f is the one I prefer, I think the best is option d.

francisco-dlp, 2022-09-12:
The problem with `io` in the logo is that it changes the "official" style of the name from RosettaSciIO to RosettaSciio. Personally, I very much prefer the former.

ericpre, 2022-09-12:
Yes, I know, but the fact that everyone (4-5 people) that I have asked read it as 10, it did put me off from option f!
As the `io` is in the stone and is distinctive from the rest of the name, I think this is fine and a good compromise.

jat255, 2022-09-12:
Based off the other discussion, I agree that lower-case is probably better for the logo. I'm partial to either *b* or *d*. One issue with *d* to me is that the two "i" characters right next to each other are slightly different size, which is a bit jarring. *b* utilizes a different font, so it's not so obvious or off-putting being a different size.

ericpre, 2022-09-12:
> One issue with _d_ to me is that the two "i" characters right next to each other are slightly different size, which is a bit jarring

Yes indeed, can it be fixed? :)

francisco-dlp, 2022-09-12:
A remark regarding lowercase `io`: that's not an acronym but a [moth](https://en.wikipedia.org/wiki/Automeris_io). Here is its chenille, that, now that we are at it, could replace the Python in the logo ;)  [image]?

jat255, 2022-09-12:
> that's not an acronym but a [moth](https://en.wikipedia.org/wiki/Automeris_io)

It's also a moon! https://en.wikipedia.org/wiki/Io_(moon)

jordiferrero, 2022-09-18:
@francisco-dlp Here is the change in logo f. The spacing between the lower case i and the higher case I is x3 that of the space between the c and the i in "Sci". I have reduced it to what it was before. The reason the spacing is not x1 is that the black stone background acts as a "character" and you need a bit of extra spacing otherwise the I cannot be read.

[image]

@jat255 @ericpre Here are versions b-d with improved spacing and sizing.
[image]

I personally prefer the lower case ones (b first, then d). I also asked some colleagues and they read 10 instead of IO...

Finally, here is a mock for the shorter versions (I don't particularly like any of them more than the previous designs)...
[image]

jlaehne, 2022-09-18:
I find it weird to have two different lower case `i` in version b) and still see the general problem with having lowercase in the name of these cenral letters.
To resolve the issue with the 10, I still tend towards e) - potentially with less thick 'serifs' or even slightly slanted ones:

[image]
(of course the snake needs to be added still)

jat255, 2022-09-18:
I quite like @jordiferrero's new version (b). Using a different font further accentuates that "io" is an important part of the package, and using a monospaced serif font goes with the "programming" theme. 

I do not care for the serif capitalized "IO" (sorry @jlaehne!)

francisco-dlp, 2022-09-19:
This is becoming the most contentious issue in HyperSpy's history...

-1 for lowercase `io`: while it solves the confusion with 10, it makes it harder to understand what `io` means. Also, I am with @jlaehne, I find it less aesthetically pleasing. Therefore, if we must solve the confusion with 10, just like @jlaehne,  +1 for version e).

Regarding short vs long version, there are multiple options:

| | Name | Library | Logo |
| - | -------- | --------- | ------- |
| a | RosettaSciIO | rosettasciio | RosettaSciIO |
| b / b2 | RosettaSciIO | rsciio / rsio | RosettaSciIO |
| c / c2 | RosettaSciIO | rsciio / rsio | RSciIO / RSIO |
| d / d2 | RSciIO / RSCI | rsciio / rsio | RSciIO / RSIO |

+1 for a) and c)

francisco-dlp, 2022-09-19:
Another for the IO in the logo is adding a dot of top of the I, like this **İ**.

ericpre, 2022-09-19:
I agree that variant e) is a good comprise:
- it is capitalised - in line with the name
- it can't be misread as 10

I was initially put off by using different fonts but as @jat255 mentioned [above](https://github.com/hyperspy/rosettasciio/issues/5#issuecomment-1250387628), it is actually useful to make it more distinctive.

Even if this is a bit tedious, it is worth taking the time to get it as good as we can from the start. In any case, this can be improved later, as this wouldn't be set in stone - pun intended! ;)

jlaehne, 2022-09-19:
I found it OK to modulate Lato, because in principle I did like Lato and there are other Sans Serif fonts with such an I, e.g. "Segoe UI", "Sen", "Amble", "Antic Sans":

[image]

But I could also live with the Code fonts you propose.

francisco-dlp, 2022-09-20:
I find that Antic Sans goes well with the ancient roots of the project.

jordiferrero, 2022-09-20:
> I found it OK to modulate Lato, because in principle I did like Lato and there are other Sans Serif fonts with such an I, e.g. "Segoe UI", "Sen", "Amble", "Antic Sans":
> 
> [image]
> 
> But I could also live with the Code fonts you propose.

Perfect. It seems we are now converging. I would suggest using the thicker fonts because (i) it works better for small logos + miniaturizations and (ii) it allows for the addition of the snake (which I am a big fan of). I like Amber more of these 3...

jlaehne, 2022-09-20:
If you want thicker fonts, I even found a few more ;-) To make the choice even harder ( I did not like the squared dot on the i for Amble):

[image]

francisco-dlp, 2022-09-21:
I like Noto's I, +1 for it.

jat255, 2022-09-21:
I prefer the straight serifs, but happy to put it to a vote. I'm good with eitehr Radio Canada or Harmattan. Here it is with some tweaked sizing on the `IO` part (`I` is exact same height as the `R` and `S`):

[image]

I guess @jlaehne and I must have used slightly different methods to vectorize the stone outline, but I prefer the slightly softer edges in my version:

This the SVG, which doesn't display properly for me in Chrome (probably since it's an obscure font -- possibly something to be aware of):
[image]

jlaehne, 2022-09-21:
I can live with both Noto or Radio Canada ;-)

> I guess @jlaehne and I must have used slightly different methods to vectorize the stone outline, but I prefer the slightly softer edges in my version:

As long as it is vectorized in the final version, I'm happy ;-) I used Inkscape, but even that has multiple options.

> This the SVG, which doesn't display properly for me in Chrome (probably since it's an obscure font -- possibly something to be aware of):

The final version should probably have the text converted to paths to avoid any font issues!

ericpre, 2022-09-22:
Before we start suffering from logo fatigue syndrome, I would suggest that we down select features with some polls.
Once we have a good idea of what features are the most popular, it will be a matter of refining the alignment, height, width, etc. 😊

I will try to select those options which seem to have been of interest - please shout if some are missing in the polls.

ericpre, 2022-09-22:
🚀 for Noto Sans Display Semi-bold
🎉 for Radio Canda Semi-bold
👀 for Lato

ericpre, 2022-09-28:
From the polls, we can narrow down to 2 candidates:
- Lato with different `IO` from Noto Sans Display Semi-bold
- Noto Sans Display Semi-bold

@jordiferrero: could please make these two (final?) candidates with the snake?

jordiferrero, 2022-10-08:
> From the polls, we can narrow down to 2 candidates:
> 
> * Lato with different `IO` from Noto Sans Display Semi-bold
> * Noto Sans Display Semi-bold
> 
> @jordiferrero: could please make these two (final?) candidates with the snake?

OK. Here are the final versions: let's make a final final poll. 🎉 --> Lato + Noto   OR   🚀--> Noto

Lato + Noto 🎉:
[image]

Noto 🚀:
[image]

ericpre, 2022-10-13:
It sounds like that RosettaSciIO now has a logo! :)
@jordiferrero, do you want to make a pull request with the exported file and raw file (for potential future change)? Thank you!