hyperspy/rosettasciio #16: Add support for the new EDAX .hd5 format

jlaehne, 2022-08-22:
As brought up on gitter by @TommasoCostanzo, it would be nice to support the new EDAX `.h5` (hdf5) file format for EDX measurements.

The main thing is that one file can contain quite a number of scans. Hierarchically, the file is divided in `samples`, then `areas` and each of them can contain multiple spectra, linescans and/or maps. So we would need a good mechanism for choosing whether to import one or several elements from the file - and which exactly - and then to correctly transform them into hyperspy object(s).

As APEX-EBSD is also using `.h5` (though the image files are saved separately in `.up2` format), `KikuchiPy` @hakonanes seems to already have a certain support for the new EDAX format? And the hierarchy should be similar between EDX and EBSD files.

reetuelzajoseph, 2023-07-25:
The latest extension is .edaxh5. The structure for EBSD looks like this similar to the way @jlaehnedescribed above:

[image]

For the EDS dataset the structure looks this:

[image]

jlaehne, 2023-07-25:
@mkuehbach is also working on reading the edaxh5 format for edx and ebsd into python in the context of https://github.com/FAIRmat-NFDI

mkuehbach, 2023-07-25:
Indeed: I have implemented Python which parses all of the data nodes in the above-mentioned EDAXH5 HDF5 file. The key question is how to share it, the much more important practical question though is what do the many terms in these fields mean. The complete glossary is available here: https://github.com/FAIRmat-NFDI/em4nfdi/tree/main/matwerk, in fact also for another dataset. Every feedback to resolve terms in the glossary via a simple text or Excel file is appreciated, ideally this should be merged back into a glossary of the EDAXH5 formats, or easier, representatives from EDAX help here to avoid such reverse engineering.
Many fields and attributes are low-level details and parameter of algorithms, some of which with unclear provenance how they end up in the EDAXH5 file and thus how trustworthy is it to have such data/metadata be consumed by research data management systems.

Three things need to happen:
- [ ] Technically read EDAXH5, for this I have code, solved problem
- [ ] Identify what fields and terms mean conceptually, this is a community effort, feel free to open issues here https://github.com/FAIRmat-NFDI/em4nfdi
- [ ] Already in parallel to identifying the meaning of terms, identify which data and metadata are relevant to be pull over for hyperspy

@reetuelzajoseph and team feel invited to contribute here.

hakonanes, 2023-07-25:
@reetuelzajoseph and @mkuehbach, great that you want to push for parsers of EBSD and orientation data in RosettaSciIO. It would be good if this work was done in conjunction with moving parsers from kikuchipy to RosettaSciIO (which I plan to start on this autumn, or, alternatively, can help people with moving then). This is to avoid duplication of efforts and that as many people as possible benefit from the developments.

We have readers for EBSD data (not orientation data) acquired on square grids (hexagonal unsupported) in the following EDAX formats in kikuchipy:
* Binary .up1 (8-bit), .up2 (16-bit). The reader is based on work by @drowenhorst-nrl for [PyEBSDIndex](https://github.com/USNavalResearchLaboratory/PyEBSDIndex)
* HDF5 ([reader](https://github.com/pyxem/kikuchipy/blob/develop/kikuchipy/io/plugins/edax_h5ebsd.py) using the [HDF5 reader base class](https://github.com/pyxem/kikuchipy/blob/develop/kikuchipy/io/plugins/_h5ebsd.py)). This reader is based mostly on files I've received from colleagues, such as @CiosG.

These are established, published readers which are actively in use and have been patched a couple times (supported).

> The key question is how to share such code and make it useful for hyperspy

My suggestion is to implement a reader here in RosettaSciIO.

> [...] following should happen for hyperspy: answer which data and metadata are relevant for hyperspy and pull them over into dictionaries

kikuchipy is HyperSpy's extension for EBSD data. The [current reader](https://github.com/pyxem/kikuchipy/blob/develop/kikuchipy/io/plugins/edax_h5ebsd.py#L73-L162) of EDAX' HDF5 files in kikuchipy reads the patterns, detector information (the pattern centers, camera tilt, sample tilt, camera azimuthal [about vertical]) the map step sizes, working distance, and magnification. I'd start with this.

kikuchipy and HyperSpy's extension for diffraction in TEM, pyxem, both use orix for handling of orientation data. There are many readers of orientation data in orix, however, a reader for orientation data from EDAX HDF5 files is missing. RosettaSciIO so far does not have any readers of orientation data. If it is decided that orientation data is within the scipe of RosettaSciIO, we should move IO from orix to RosettaSciIO. Again, coordinating efforts would be best. Readers in orix should return dictionaries of arrays and such, but as of now they return a `CrystalMap` class instance, which is similar to MTEX' `EBSD` class. A `CrystalMap` expects [this information](https://orix.readthedocs.io/en/stable/reference/generated/orix.crystal_map.CrystalMap.html). I'd start with this.

CSSFrancis, 2023-07-26:
We probably need to think a little bit more about what happens when a signal doesn't fit nicely into a hyperspy signal.  I think that we should support every file format that people are willing to add, but we probably need to just add a flag that this file format isn't supported by hyperspy (and maybe a suggestion of what software to use).

@jlaehne Do you have any thoughts on that?

hakonanes, 2023-07-27:
> We probably need to think a little bit more about what happens when a signal doesn't fit nicely into a hyperspy signal

This discussion probably deserves its own issue?

ericpre, 2023-07-28:
Thank you @reetuelzajoseph and @mkuehbach for your comments.

> I have implemented Python code which parses all of the data nodes in the above-mentioned EDAXH5 HDF5 file(s).
The key question is how to share such code and make it useful for hyperspy @ericpre

@mkuehbach, as already mentioned by @hakonanes, rosettatsciio will be a very good place to share such a reader and will be readily available to hyperspy users or other libraries using hyperspy - rosettasciio will be start to be used in hyperspy 2.0, which still need to be released, but there is good progress on that front!
If you are interested to do that, that would be great and please provide feedback: as most of the readers have growth organically over the years, which overall has been very useful and successful, there is quite a lot of room for improvement on the consistency between readers, documentation, on the way things are being done, etc. 

> The much more important practical question though is what do the many terms in these fields mean: The complete glossary as based on examples is parsed out here: https://github.com/FAIRmat-NFDI/em4nfdi
Every community feedback to resolve terms in the glossary via a simple text or Excel file feedback is appreciated, especially to clarify units etc. Ideally this should be merged back into a glossary of the EDAXH5 formats, or easier, representatives from EDAX could help here to avoid such reverse engineering.
>
> @ericpre Do you think there is a better place where such a collection of reverse engineered Rosetta stone for EDAXH5 could be stored? If so we should have a meeting with @reetuelzajoseph to make this happen.

We started to collect [known information](https://rosettasciio.readthedocs.io/en/latest/file_specification/index.html) about file formats/specifications to make it more accessible but there was little effort on this and this is something that we should enforce it, particularly when implementing new formats/functionalities. Quite often, something is implemented using reference materials available online and at some point, this materials is not available anymore... so we should make sure that we don't lose this information.
Of course, there is a lot of information within the reader themselves, but this is not easy to read or poorly documented and we should have an documentation of the file format itself as part of the documentation. This would help a lot with maintenance! As this should be done at the same time as new functionalities/formats are implemented, I think that it would make sense that it is added to the rosettasciio documentation and this is something that the community could help with during review, etc.

To give more context on rosettasciio:
- the IO code has been split recently from hyperspy to make with the aim more accessible to other libraries/frameworks.
- In term of metadata handling, the current approach is pragmatic, in a sense that we try to read all possible metadata as _they come_ into the `original_metadata` and parse some of them to `metadata` into a [tree structure](https://hyperspy.org/hyperspy-doc/dev/reference/metadata.html), which has been defined in ad-hoc fashion. This is working but not very well and this is something that will need improvement sooner than later, because this is still very hyperspy centric and hyperspy has its own set of metadata, that is used for various functionalities. Hopefully, at some point, we can start to use an established standard and ongoing effort by other initiatives should help with this! 😃 
- The public API is currently very minimal, but as the community grows and we gain better understanding on what API would be useful, it is likely that the public API will be extended.

ericpre, 2023-07-28:
> > We probably need to think a little bit more about what happens when a signal doesn't fit nicely into a hyperspy signal
> 
> This discussion probably deserves its own issue?

Yes, can you please open an separate issue? You will have a better idea than me on this to describe the needs, whether it would make sense or not, etc.
Afew readers read simple process data (for example, EDS map) but orientation data may have structure which is more complicated than that - I don't know! 😉

mkuehbach, 2023-07-28:
For the most general case orientation data are de facto point cloud with mark data. For each point in R^2 or R^3 (provided some serial-sectioning reconstruction was performed) there are associated mark data, that is phase (an id from a dictionary of phases against one indexed or e.g. a marker notIndexed), orientation value tuple (based on parameterization), and pattern and eventually indexing quality descriptors (bc, ci, etc.), 

From what I have seen in the literature https://github.com/FAIRmat-Experimental/nexus-fairmat-proposal/blob/main/9636feecb79bb32b828b1a9804269573256d7696/_sources/classes/contributed_definitions/NXem_ebsd.rst.txt is the most complete and systematic approach.

With AI/ML-based indexing solutions between simulated and measured patterns can give probabilistic/fuzzy descriptors for the match, this is different to traditional approaches where e.g. confidence index whereby essential a boolean answer (indexed or not) is produced. When I wrote NXem_ebsd I drafted it so general in that most people can find themselves. Ignore for now all existence constraints there but already the taxonomy which is one part of the NXem_ebsd data schemas give a complete view about how EBSD data in principle can be pumped out by software (open source or technology partner)

ericpre, 2023-08-03:
@mkuehbach, this should released between mid- and end of September.