hyperspy/rosettasciio #36: SFS Reader Origins and license.

imikejackson, 2022-09-10:


ericpre, 2022-09-10:
Transfered to [hyperspy/rosettasciio](https://github.com/hyperspy/rosettasciio) as the IO development will continue here.

sem-geologist, 2022-09-14:
@imikejackson , as Your question got answered here, would You mind closing this issue? BTW, the RE of that format took me few years, but that was worth as it opened the format and is possible to offload processing of bcf's offline to servers with any OS.
If You would want any help with EBSD RE I could share hints and mentor if need that, I had looked initially into that (our lab has Brukers EBSD), the raw data is packed very similarly compared to binary data of EDS cube, and Esprit interpreted data is very easy to get as those are xml. I however have no more time for doing anything with EBSD, I work full-time with EPMA, thus had never implemented the reader.

vasole, 2022-10-10:
Concerning the license of the code, I doubt you can legally change it without the agreement of the original author @sem-geologist

sem-geologist, 2022-10-10:
@vasole what project Are You talking about?

vasole, 2022-10-10:
This one. My comment is related to #51 Unless you made a official copyright transfer, if you released the code under GPL, only you can change the license.

sem-geologist, 2022-10-10:
why so pessimistic? It was already discussed before at HyperSpy, and now as io library is going to be split from HyperSpy I think it is going even easier to change that license (At least I would agree it to be re-licensed to LGPL or BSD). Are Your project in particularly relying on python solution?

vasole, 2022-10-10:
It's not about being pessimistic. We changed the license of the fabIO project and we had to contact and get the agreement of all the contributors to do so. So, things can be cumbersome.

My project is the XRF analysis code PyMca (https://github.com/vasole/pymca), and what some of my users need is SFS support. That is the short term need.

For the long run, I guess not having to maintain software support of multiple file formats in my code and being able to rely on (and contribute to) a widely maintained library can only be of benefit to everybody.

closed 2025-03-23 by imikejackson