hyperspy/rosettasciio #38: Consolidate docstring and documentation

ericpre, 2022-09-12:
The problem is also, that it is not consistent how detailed the `file_reader` and `file_writer` docstrings are. For some formats more is documented in the user guide, for others more in the docstrings.

_Originally posted by @jlaehne in https://github.com/hyperspy/rosettasciio/pull/35#discussion_r968012172_

ericpre, 2022-09-17:
One approach could be to:
- put the generic/introductory plugin specific documentation a `rst`/`md` file for each plugin, which could live in the `docs` folder or in the plugin folder
- add the `file_reader` and `file_write` docstrings in the plugin specific page of the docs

This could be done by one of the two:
- use a sphinx template
- include the API reference in the plugin specific documentation page.

Considering that sphinx is a bit brittle and not always clearly documented, using the sphinx template approach may be more difficult but it would have the advantage of enforcing consistence between plugins and avoid repeating code.

jlaehne, 2022-09-17:
>     * put the generic/introductory plugin specific documentation a `rst`/`md` file for each plugin, which could live in the `docs` folder or in the plugin folder

That file exists for every plugin afaik, in `docs/supported_formats/`.
 
>     * add the `file_reader` and `file_write` docstrings in the plugin specific page of the docs

That remains to be done, though some plugins have very detailed docstrings and others none at all.

PR #76 (2023-01-22): Doc and import improvement #5
### Description of the change
Follow up on #47, #59, #64, and #72. Closes #38. For readers with folder names from D...H. Final PR of this series.

### Progress of the PR
- [x] Change implemented (can be split into several points),
- [x] update docstring (if appropriate),
- [x] update user guide (if appropriate),
- [x] add an changelog entry in the `upcoming_changes` folder (see [`upcoming_changes/README.rst`](https://github.com/hyperspy/rosettasciio/blob/main/upcoming_changes/README.rst)),
- [x] Check formatting changelog entry in the `readthedocs` doc build of th [... cut]
closed 2023-01-22 by ericpre