hyperspy/rosettasciio #50: [QUESTION] Any chance to make SFS reading code LGPL?

vasole, 2022-10-10:


ericpre, 2022-10-10:
@vasole, I transfer the issue to this repository as we have split the IO part of hyperspy. RosettaSciIO is not released yet but we aren't very far to make the first release. Would it work for you to use RosettaSciIO as dependency?

Regarding changing the license to a more permissive is a separate discussion! I have open https://github.com/hyperspy/rosettasciio/issues/51, you are welcome to make a case for LGPL! :)

vasole, 2022-10-10:
Typically what I do for "small" things not widely distributed (i.e. missing in Debian) is to try to import them from the real thing (RosettaSciIO would be in this case) and, if not available, revert to a minimalist local copy of what I need.

Putting a dependency in HyperSpy would have meant to ship a a lot of additional stuff when building frozen binaries and it was clearly not an option.

If RosettaSciIO is reasonable in its dependencies, it could simplify my work since I am also maintaining quite a few formats for spectroscopy. If its license is GPL, it would be a no-go.

I close the issue since it is somehow overwritten by #51.

closed 2022-10-10 by vasole

ericpre, 2022-10-10:
> Putting a dependency in HyperSpy would have meant to ship a a lot of additional stuff when building frozen binaries and it was clearly not an option.
>
> If RosettaSciIO is reasonable in its [...]

Yes, this was the one of the motivation to split the IO part!

Does it the current dependency requirement sound reasonable? We could easily make h5py, numba, sparse and imageio optional, if this is useful. https://github.com/hyperspy/rosettasciio/blob/8176d44373a6173059863e9b7e06cca923fcb74c/setup.py#L154-L166

vasole, 2022-10-10:
I do not know all the formats you have in mind nor the data layout you intend to present to the user, therefore I have difficulties to understand the need of using pint, scipy, dask,... 

I would distinguish between reading a file format and reading a file format in view of a particular target. For instance, you might be reading the data in view of its use by HyperSpy instead of just reading the data without any abstraction. I do not know if it is clear what I mean.

For instance, in our silx-kit/silx project, we have an abstraction to access all supported file formats as if they would have converted to NeXus/HDF5. However, the pieces dealing with the different file formats (fabIO, specfile, h5py,...) know nothing about NeXus. It is an intermediate layer (in silx) that takes care of it.

As I said, I do not know if RosettaSciIO is aiming at providing access to the content of the files, with an additional layer to tailor them to HyperSpy or if the readout is already with a target in mind.

ericpre, 2022-10-10:
I see what you mean and it does make sense. The code has grown in a ah-hoc fashion with community contributions. It follows a pattern where the reader are separated from the rest of the code but its separation hasn't been strongly enforced. Indeed, it would make scipy and pint optional, as they are used for unit conversion or interpolation, that can done after reading. Regarding dask, it would be more difficult as it is used to load data lazily.