hyperspy/rosettasciio #56: Jeol.py error when loading asw files containing sequential acquisition

PierFF, 2022-10-29:
Hi everyone,
I get the following error "_read_eds() takes 1 positional argument but 2 were given" when using the jeol.py io plugin on asw files (such as this one [1.zip](https://github.com/hyperspy/hyperspy/files/9879857/1.zip)). After a quick glance at the code I have the impression that reading sequential acquisition is not supported. Can someone confirm that ?
Is someone else interested in this feature ?
Since I would like to use this feature, has anyone any idea of how the region associated to the eds file can be identified in the asw tag structure returned by the _parsejeol function ? Could it be associated withe the PositionMM tag associated to the enumerated entries in the ViewData structure ? (to be honest I do not fully understand what the PosiitonMM and PositionMM2 arrays represent...)

Thanks in advance for your help,

Pier

ericpre, 2022-10-29:
I transferred the issue to the [hyperspy/rosettasciio](https://github.com/hyperspy/rosettasciio) repository as the IO development is now taking place here - see https://github.com/hyperspy/hyperspy/issues/1978 for context.

> After a quick glance at the code I have the impression that reading sequential acquisition is not supported. Can someone confirm that ?

I don't remember anything about sequential reading, so I don't think that this is implemented.

> Is someone else interested in this feature ?

Usually, there is always someone who is interested! ;)

> Since I would like to use this feature, has anyone any idea of how the region associated to the eds file can be identified in the asw tag structure returned by the _parsejeol function ? Could it be [...]

@nem1234 and @arnduveg are the main contributors to the jeol reader. From what I remember, @arnduveg has finished their PhD and is most likely not working on this anymore but @nem1234 may be able to help and will be the one who knows best.

nem1234, 2022-10-31:
Thank you reporting the problem in jeol EDS reader.

Ok, I recognized the problem.
Currently, the positional parameters are not decodeded in the code.

I'll try to fix the problem, but I think it takes several days to decode the positional parameters.

nem1234, 2022-11-02:
Now, I'm preparing PR.
Reading multiple eds files and putting the positionMM2 parameters on each spectrum are finished.
May be the rectangle positional marker should be put on the index image, but in such case, can I call the hyperspy's marker functions?
Otherwise I need to write codes to access metadata/Markers dict directly as in the same way of bruker/api.py.

PierFF, 2022-11-07:
Great work...thanks!!!

ericpre, 2022-11-12:
> May be the rectangle positional marker should be put on the index image, but in such case, can I call the hyperspy's marker functions?

Yes, would it make sense to add it as [permanent markers](https://hyperspy.readthedocs.io/en/latest/user_guide/visualisation.html#permanent-markers) to the corresponding image? If I understand correctly, the spectrum doesn't have a navigator in hyperspy sense (to pointer a pointer), but in the file, there will be a survey image, where the markers could be added onto.