hyperspy/rosettasciio #69: set_log_level for rosettasciio is needed

nem1234, 2022-11-28:
#### Describe the functionality you would like to see.
Like hyperspy's  hs.set_log_level function, rosettasciio.set_log_level is needed.
May be hs.set_log_level should also control rsciio.set_log_level.

ericpre, 2022-11-29:
> Like hyperspy's hs.set_log_level function, rosettasciio.set_log_level is needed.

Yes, indeed it would be convenient to have something similiar.

> May be hs.set_log_level should also control rsciio.set_log_level.

In principle, it should stay separated.

nem1234, 2022-11-30:
> In principle, it should stay separated.
I agree it, but, currently no warning messages are shown while loading files from hyperspy.
Setting default logging level of WARNING seems useful.

closed 2023-12-06 by jlaehne