hyperspy/rosettasciio #91: Incorrect Scale Conversion for Non-FEI MRC Files

AndrewHerzing, 2023-03-08:
For MRC files that are not collected by FEI software, the scale is assumed to be in Angstroms and Hyperspy converts it to nanometers. However, the conversion is done by multiplying by 10 rather than dividing by 10.  The relevant lines of code are 179 - 187 in `mrc.py`.
```python
    if fei_header is None:
        # The scale is in Amstrongs, we convert it to nm
        scales = [10 * float(std_header['Zlen'] / std_header['MZ'])
                  if float(std_header['MZ']) != 0 else 1,
                  10 * float(std_header['Ylen'] / std_header['MY'])
                  if float(std_header['MY']) != 0 else 1,
                  10 * float(std_header['Xlen'] / std_header['MX'])
                  if float(std_header['MX']) != 0 else 1, ]
        offsets = [10 * float(std_header['ZORIGIN']),
                   10 * float(std_header['YORIGIN']),
                   10 * float(std_header['XORIGIN']), ]
````

CSSFrancis, 2023-03-08:
Hi @AndrewHerzing! It might be better to have this as a rosettasciio issue rather than a hyperspy issue.  We can always make the changes both places but it is just better to have all of the discussion in rosettasciio rather than in hyperspy.

AndrewHerzing, 2023-03-09:
This makes sense.  Looks like a pretty easy fix.

AndrewHerzing, 2023-03-10:
Here is a link to test dataset which will reproduce the error:

https://drive.google.com/file/d/1zv1gaa3YYe8Sg5kbaUsPx3qV5yfcFsyX/view?usp=sharing

It is a short tilt series consisting of 9 images which are 512x512 pixels each with 'int16' data type.  The pixel size should be 4.416 nm rather than 441.6 as the reader is currently written.

ericpre, 2023-06-04:
@AndrewHerzing, thank you for sharing the file, we are finally getting around sorting these mrc issues in #131. Is it possible to get smaller test file? If so, when you have a chance, could get and share a smallest possible file to add to the test suite please? Was the file created using IMOD/SerialEM?

ericpre, 2023-06-04:
This was close automatically when merging #131, but let's keep it open for now so that we don't forget to add some test file.

AndrewHerzing, 2023-06-05:
Here is a ~1MB MRC that was generated with SerialEM.  This is the smallest I can create at the moment because our scope is down for repair.  The scale as read by Hyperspy is 315 nm whereas it should be 3.15 nm.

https://drive.google.com/file/d/1vGQO9VM1JFyOwHMzZYnDxnnPdejTDw58/view?usp=sharing

jlaehne, 2023-06-05:
By the way, with the development versions of HyperSpy (RELEASE_next_major branch) and RosettaSciIO, you should already be able to read the files correctly - in case you are comfortable to install these.

ericpre, 2023-06-06:
Thank you @AndrewHerzing, would it be possible to wait to even smaller file? Maybe with different dtype too? The bug has been fixed and until we have a new file we can check manually, when necessary. it seems that most of the code was more than 10 years for the mrc reader (possibly written at a time, where writing test wasn't enforced!) and it is not like the code need to be changed very often! :)