hyperspy/rosettasciio #92: Add a function of saving file as Gatan's dm3/dm4 file

Leexulingyi, 2023-03-10:
#### Describe the functionality you would like to see.
 Why Gatan's dm3/dm4 write function is not completed since their read function is. Is feasible of adding it?
We're trying to develop the write function of Gatan's dm3/dm4 files, by reading the source code.
What kinds of problems will we face?

jlaehne, 2023-03-10:
Gatan's format is an undocumented, binary format. Reading functionality was developed over the years by reverse engineering. However, as long as Gatan does not publish the specifications of their file format, these functions will never be complete and it will not be possible to provide writing to that file format. Besides the technical challenge, focus of the developers will always be in support for writing to open formats that can be more widely used.

However, if you want to transfer data from python/HyperSpy to DigitalMicrograph, there is a number of strategies based on open formats documented in the user guide: https://rosettasciio.readthedocs.io/en/latest/interoperability.html So there should be no need to implement a file writer.

hakonanes, 2023-03-10:
> We could probably also improve our dm3 reader by copying theirs.

There has been some talk in issues about using a more permissive license than GPL for parts of HyperSpy. Don't know whether anything of the current DM3 plugin is derived from work under a GPL license, but if not, and there is serious interest in moving towards a more permissive license, copying from Nion Swift forces a GPL license on the current plugin. Something to be aware of.

CSSFrancis, 2023-03-10:
[Here](https://github.com/nion-software/nionswift-io/tree/master/nionswift_plugin/DM_IO) is the actual repo if you want to look at it.  It has a Apache2.0 license which doesn't really mean much to me.  If we use their schema but then write our own code is that acceptable?  I'm not sure how this exactly works.

hakonanes, 2023-03-10:
People in [this Stack Overflow thread](https://softwareengineering.stackexchange.com/questions/40561/is-bsd-license-compatible-with-apache) claim the Apache 2.0 license is compatible with more permissive licenses like BSD/MIT. However, [SciPy (BSD)](http://scipy.github.io/devdocs/dev/hacking.html#license-considerations) does not accept code that is derived from Apache licensed code. So I'm not sure. And sorry, I only looked at the Nion Swift library (GPL) and assumed wrongly the plugin was then GPL.

CSSFrancis, 2023-03-10:
> People in [this Stack Overflow thread](https://softwareengineering.stackexchange.com/questions/40561/is-bsd-license-compatible-with-apache) claim the Apache 2.0 license is compatible with more [...]

I was on github mobile so I couldn't find where Nion actually had their i-o code so my first link wasn't very helpful :)  The Apache 2.0 license has stricter language regarding patent infringement and offers slightly better protection there which makes sense why Nion chose it.  It does say that it isn't strictly [compatible](https://en.wikipedia.org/wiki/Apache_License#Reception_and_adoption) with BSD.