hyperspy/rosettasciio #99: Initial Release

jlaehne, 2023-04-17:
## [v0.1](https://github.com/hyperspy/rosettasciio/milestone/1)
We have progressed quite far on preparing an initial release and improving consistency within RosettaSciIO as documented in the [milestone](https://github.com/hyperspy/rosettasciio/milestone/1). Therefore, we should aim for a timely initial release. Let's try to use this issue to speed up that process and see what is left. In principle, I think we could even release the current state.

However, it would be good to clarify the following points:
- [ ] #51
- [ ] #14 (currently the test data blows up the project size)
- [ ] #69 
- [ ] #79

Finally, there are a number of reported bugs that would be nice to get fixed (and most of them should be low hanging fruits):
- [ ] #37
- [ ] #66 
- [ ] #71
- [ ] #73
- [ ] #82
- [ ] #83 
- [ ] #90
- [ ] #91 
- [ ] #96

And of course, it is always good to
- #60

**Once, we are happy with the state, we need to**
- [ ] Document releasing workflow
- [ ] Create RosettaSciIO on pip
- [ ] Do initial release on github
- [ ] Create conda-forge feedstock

jlaehne, 2023-04-17:
Do feel free to edit and add to the above list.

ericpre, 2023-04-18:
We could merge the existing the PR ready to review and proceed with the release. The rest can be addressed in following releases.

ericpre, 2023-06-01:
After #129 and #131 are done and merged, I think that we can proceed with the release, everything should be in place.

jlaehne, 2023-06-02:
#55 and #87 are also ready for review and may be included in the initial release - though they should not hold back the release as we can then easily do follow-up releases.

ericpre, 2023-06-02:
#11 is quite large and may take time to review - except it someone already started reviewing it?

CSSFrancis, 2023-06-06:
I need to make some changes to #11 based on the changes introduced by #111 so I was kind of waiting before going back to it.  I would vote for going for the 0.1.0 release and then we can add it later.

jlaehne, 2023-06-07:
The initial release is out: #133

closed 2023-06-07 by jlaehne