hyperspy/rosettasciio #115: Error when trying to load JEOL's .map file

PereraUmesha, 2023-05-08:
#### Describe the bug
I am trying to load a .map file from JEOL's EPMA (Electron Probe Micro Analyzer).
Then I get the below error.

WARNING:hyperspy.io_plugins.jeol:Not a valid JEOL img format
ERROR:hyperspy.io:If this file format is supported, please report this error to the HyperSpy developers.
---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
Input In [4], in <cell line: 1>()
----> 1 mdata = hs.load(map_fname)

File ~\anaconda3\envs\new202302\lib\site-packages\hyperspy\io.py:466, in load(filenames, signal_type, stack, stack_axis, new_axis_name, lazy, convert_units, escape_square_brackets, stack_metadata, load_original_metadata, show_progressbar, **kwds)
    463         objects.append(signal)
    464 else:
    465     # No stack, so simply we load all signals in all files separately
--> 466     objects = [load_single_file(filename, lazy=lazy, **kwds)
    467                for filename in filenames]
    469 if len(objects) == 1:
    470     objects = objects[0]

File ~\anaconda3\envs\new202302\lib\site-packages\hyperspy\io.py:466, in <listcomp>(.0)
    463         objects.append(signal)
    464 else:
    465     # No stack, so simply we load all signals in all files separately
--> 466     objects = [load_single_file(filename, lazy=lazy, **kwds)
    467                for filename in filenames]
    469 if len(objects) == 1:
    470     objects = objects[0]

File ~\anaconda3\envs\new202302\lib\site-packages\hyperspy\io.py:525, in load_single_file(filename, **kwds)
    518     raise ValueError(
    519         "`reader` should be one of None, str, "
    520         "or a custom file reader object"
    521     )
    523 try:
    524     # Try and load the file
--> 525     return load_with_reader(filename=filename, reader=reader, **kwds)
    527 except BaseException:
    528     _logger.error(
    529         "If this file format is supported, please "
    530         "report this error to the HyperSpy developers."
    531     )

File ~\anaconda3\envs\new202302\lib\site-packages\hyperspy\io.py:545, in load_with_reader(filename, reader, signal_type, convert_units, load_original_metadata, **kwds)
    543 """Load a supported file with a given reader."""
    544 lazy = kwds.get('lazy', False)
--> 545 file_data_list = reader.file_reader(filename, **kwds)
    546 signal_list = []
    548 for signal_dict in file_data_list:

File ~\anaconda3\envs\new202302\lib\site-packages\hyperspy\io_plugins\jeol.py:109, in file_reader(filename, **kwds)
    107         fd.close()
    108 else:
--> 109     d = extension_to_reader_mapping[file_ext](filename, **kwds)
    110     if isinstance(d, list):
    111         dictionary.extend(d)

File ~\anaconda3\envs\new202302\lib\site-packages\hyperspy\io_plugins\jeol.py:194, in _read_img(filename, scale, **kwargs)
    190     _logger.warning("Not a valid JEOL img format")
    192 fd.close()
--> 194 return dictionary

UnboundLocalError: local variable 'dictionary' referenced before assignment

#### To Reproduce
Steps to reproduce the behavior:
```
import hyperspy.api as hs
from pathlib import Path

map_fname = Path("data/inputdata/data001.map")
mdata = hs.load(map_fname)
```

#### Expected behavior
Load the contents of .map file

#### Python environement:
 - HyperSpy version: 1.7.3
 - Python version: 3.9.16

#### Additional context

ericpre, 2023-05-08:
It seems that this file is not supported, can you please share a file that we can use to reproduce and test?

PereraUmesha, 2023-05-08:
Thank you for the response. 
Please find the file (Test.map) inside the following .zip file

[Test.zip](https://github.com/hyperspy/rosettasciio/files/11425855/Test.zip)

nem1234, 2023-05-15:
Is the size of image fixed with 256x256 pixels?
Is there another file that was saved with *.img file?

nem1234, 2023-05-15:
This test file contains 32 bytes header: 
00000000  01 00 00 00 00 00 00 00  00 00 00 00 08 00 00 00  |................|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|

And also 256x256 pixels double precision float values (8bytes / pixel).
00000020  00 00 00 00 00 cc b1 40  00 00 00 00 00 c0 b1 40  |.......@.......@|
....

nem1234, 2023-05-25:
We need more sample data to resolve this issue.
Do you have some related metadata files with this map?

ericpre, 2023-07-08:
@PereraUmesha, would you be able to provide the information asked by @nem1234? Thanks!