hyperspy/rosettasciio #127: loading 4D STEM data (.mrc files) generated by TFS Velox using Ceta camera

yuanh9, 2023-05-27:
Hyperspy complains about loading .mrc files (4D STEM dataset generated by TFS Velox using Ceta camera):

WARNING:hyperspy.io_plugins.mrc:There was a problem reading the extended header
ERROR:hyperspy.io:If this file format is supported, please report this error to the HyperSpy developers.
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[2], line 1
----> 1 s=hs.load("E:/test/test.mrc")

File ~\.conda\envs\hyperspy_env\lib\site-packages\hyperspy\io.py:466, in load(filenames, signal_type, stack, stack_axis, new_axis_name, lazy, convert_units, escape_square_brackets, stack_metadata, load_original_metadata, show_progressbar, **kwds)
    463         objects.append(signal)
    464 else:
    465     # No stack, so simply we load all signals in all files separately
--> 466     objects = [load_single_file(filename, lazy=lazy, **kwds)
    467                for filename in filenames]
    469 if len(objects) == 1:
    470     objects = objects[0]

File ~\.conda\envs\hyperspy_env\lib\site-packages\hyperspy\io.py:466, in <listcomp>(.0)
    463         objects.append(signal)
    464 else:
    465     # No stack, so simply we load all signals in all files separately
--> 466     objects = [load_single_file(filename, lazy=lazy, **kwds)
    467                for filename in filenames]
    469 if len(objects) == 1:
    470     objects = objects[0]

File ~\.conda\envs\hyperspy_env\lib\site-packages\hyperspy\io.py:525, in load_single_file(filename, **kwds)
    518     raise ValueError(
    519         "`reader` should be one of None, str, "
    520         "or a custom file reader object"
    521     )
    523 try:
    524     # Try and load the file
--> 525     return load_with_reader(filename=filename, reader=reader, **kwds)
    527 except BaseException:
    528     _logger.error(
    529         "If this file format is supported, please "
    530         "report this error to the HyperSpy developers."
    531     )

File ~\.conda\envs\hyperspy_env\lib\site-packages\hyperspy\io.py:545, in load_with_reader(filename, reader, signal_type, convert_units, load_original_metadata, **kwds)
    543 """Load a supported file with a given reader."""
    544 lazy = kwds.get('lazy', False)
--> 545 file_data_list = reader.file_reader(filename, **kwds)
    546 signal_list = []
    548 for signal_dict in file_data_list:

File ~\.conda\envs\hyperspy_env\lib\site-packages\hyperspy\io_plugins\mrc.py:155, in file_reader(filename, endianess, **kwds)
    153 else:
    154     _logger.warning("There was a problem reading the extended header")
--> 155     f.seek(1024 + std_header['NEXT'])
    156     fei_header = None
    157 NX, NY, NZ = std_header['NX'], std_header['NY'], std_header['NZ']

TypeError: only integer scalar arrays can be converted to a scalar index

ericpre, 2023-05-27:
Would you be able to generate as small as possible test data? Let's say a series of 2 images with the highest binning possible?

yuanh9, 2023-05-27:
This dataset was generated by the microscope. I have some in smaller size. Would that help to debug?

ericpre, 2023-05-28:
Sorry, I should have been more explicit on why I was asking this. We use smallest possible test data to add to the repository and run test against it to make sure that there works fine in the future (no regression is introduce in the future). As we have many of these test data to cover as many cases as possible across file format, it is convenient to have it as small as possible to avoid growing the repository size unnecessarily. It means that these test data are acquired using the settings mentioned above (in this case, I guess: 2 images, highest binning possible) and doesn't need to contain information, noise/not signal is enough.

> This dataset was generated by the microscope. I have some in smaller size. Would that help to debug?

I guess this would be useful to figure out this specify bug but it will not be suitable to add to the repository, which will be good to prevent future regression.

yuanh9, 2023-05-28:
Thanks for you explanation. I'll generate some data like this once I'm on the microscope next week.

closed 2023-06-02 by ericpre