hyperspy/rosettasciio #128: Add information on providing small test dataset

ericpre, 2023-05-29:
Sorry, I should have been more explicit on why I was asking this. We use smallest possible test data to add to the repository and run test against it to make sure that there works fine in the future (no regression is introduce in the future). As we have many of these test data to cover as many cases as possible across file format, it is convenient to have it as small as possible to avoid growing the repository size unnecessarily. It means that these test data are acquired using the settings mentioned above (in this case, I guess: 2 images, highest binning possible) and doesn't need to contain information, noise/not signal is enough.

> This dataset was generated by the microscope. I have some in smaller size. Would that help to debug?

I guess this would be useful to figure out this specify bug but it will not be suitable to add to the repository, which will be good to prevent future regression.

_Originally posted by @ericpre in https://github.com/hyperspy/rosettasciio/issues/127#issuecomment-1565928324_

sem-geologist, 2023-06-07:
I just want to add some observation, I just checked the release size is 80 MiB (83MB in zenodo).
There are 2 dominating blowers to this size considering test files:
1. edax_files.zip           34.8 MiB
2.  4DSTEMscan.mrc   32.2 MiB

but this is unfair comparisson of these two, as mrc gets compressed down to 14MiB, and edax_files.zip can't, as it is compressed anymore. So those two files takes more than a half of release size!

ericpre, 2023-06-07:
Yes, the release on github and the zenodo entry contains the whole source code and this is good for archive purpose. On pypi, the [packages are 0.5 to 1MB](https://pypi.org/project/rosettasciio/#files) depending on the binary.
It doesn't matter how large are the zip or tarball on github and zenodo because these are not used for distribution purpose.

closed 2024-02-28 by jlaehne