hyperspy/rosettasciio #141: Ragged dtype=object `BaseSignal` saving `.hspy` should give better error

magnunor, 2023-07-05:
When trying to save a ragged `BaseSignal` with `dtype=object`, the error message is confusing for users. Example:

```python
import numpy as np
import hyperspy.api as hs

data = np.empty((3, 5), dtype=object)
for ix in range(data.shape[-1]):
    for iy in range(data.shape[-2]):
        data[iy, ix] = np.random.randint(1, 100, size=(5, 2))

s = hs.signals.BaseSignal(data, ragged=True)
s.save("test.hspy", overwrite=True)
```

Gives the error message `TypeError: Object dtype dtype('O') has no native HDF5 equivalent`. This is due to `h5py` not supporting `dtype=object`.

However, the error message could be improved to something like: `Saving ragged signals with dtype=object is not supported, try saving it as the .zspy format instead.`

ericpre, 2023-07-08:
The issue is rather that `hspy` should be able to save this type of data. A workaround may be to unfold the signal when writing (write with navigation dimension 1) and fold it to the correct navigation shape when reading - see https://github.com/h5py/h5py/issues/876.

ericpre, 2023-12-16:
Fixed in https://github.com/hyperspy/rosettasciio/pull/164.

closed 2023-12-16 by ericpre