hyperspy/rosettasciio #167: pantarhei metadata handling

EnigmaticEureka, 2023-10-05:
in pantarhei/_api.py line 344:
The filter aperture can also be "Out", meaning:
mapped.set_item(
            "Acquisition_instrument.TEM.Detector.EELS.aperture_size", float(aperture)
            )
cannot convert "Out" to a float.

Therefore I suggest to change:

if "filter.aperture" in meta_data:
        aperture = meta_data["filter.aperture"]
        if "mm" in aperture:
            aperture = aperture.split("mm")[0]
            aperture = aperture.rstrip()
          mapped.set_item(
          "Acquisition_instrument.TEM.Detector.EELS.aperture_size", float(aperture)
          )

to this:

    if "filter.aperture" in meta_data:
        aperture = meta_data["filter.aperture"]
        if "mm" in aperture:
            aperture = aperture.split("mm")[0]
            aperture = aperture.rstrip()
            mapped.set_item(
            "Acquisition_instrument.TEM.Detector.EELS.aperture_size", float(aperture)
            )
        else:
            mapped.set_item(
            "Acquisition_instrument.TEM.Detector.EELS.aperture_size", aperture
            )

This works, however, I am unaware of any other possible outputs for the aperture which may yet require more specific handling. Thank you for your consideration.

#### Python environement:
 - RosettaSciIO version: 0.1

ericpre, 2023-10-05:
Would you be interested to make a pull request to propose the changes?

EnigmaticEureka, 2023-10-16:
I believe the pull request is complete, sorry it took so long, was my first
one ever. Feel free to let me know if something is unclear.

ericpre, 2023-12-02:
Done in #173.