hyperspy/rosettasciio #184: setuptools_scm missing for some linux install, due to dist-packages location

magnunor, 2023-11-19:
If `rosettasciio` is installed via PyPi with root in (for example) Ubuntu, it is placed in `/usr/local/lib/python3.10/dist-packages`.

This leads to a problem with importing `rosettasciio`, due to  `rosettasciio` thinking it is in an editable installed (https://github.com/hyperspy/rosettasciio/blob/main/rsciio/__init__.py#L25), excepting the `setuptools_scm` package. However, the `setuptools_scm`  is not installed by default, leading to an error. For example in the Atomap testing system: https://gitlab.com/CSSFrancis/atomap/-/jobs/5567299422#L3567

I guess this is will not occur for very many people, but the fix should be fairly straight forward. For example by changing 

```python
if Path(__file__).parent.parent.name == "site-packages":  # pragma: no cover
```
to

```python
if Path(__file__).parent.parent.name in ["site-packages", "dist-packages"]:  # pragma: no cover
```

# To Reproduce

```
docker run -it ubuntu:latest
apt update
apt install python3-numpy python3-pip
pip install hyperspy==2.0rc0
python3
import rsciio
```

ericpre, 2023-11-19:
Do you want to make a R?

magnunor, 2023-12-01:
Should be resolved in https://github.com/hyperspy/rosettasciio/pull/187#event-11123557845