hyperspy/rosettasciio #212: Texts marker with navigation dimension gives error when saving

magnunor, 2024-01-19:
When making a `Texts` marker with navigation dimensions, and adding to a signal with navigation dimensions. An error is raised when trying to save it (plotting it works fine):

## Minimal working example

```python
import numpy as np
import hyperspy.api as hs
from hyperspy.drawing._markers.texts import Texts

s = hs.signals.Signal2D(np.zeros((2, 100, 100)))
text_offset = np.empty(2, dtype=object)
text = np.empty(2, dtype=object)

for i in range(2):
    text_offset[i] = [[50, 40], ]
    text[i] = ['(30.0, -15.0)']

marker_texts = Texts(offsets=text_offset, texts=text, color="red")
s.add_marker(marker_texts, permanent=True)
s.save("testsignal.hspy")
```

Gives the error:

```python
rsciio/hspy/_api.py:121, in HyperspyWriter._get_object_dset(group, data, key, chunks, **kwds)
    119     dtype = data[test_ind].compute().dtype
    120 else:
--> 121     dtype = data[test_ind].dtype
    122 dset = group.require_dataset(
    123     key, data.shape, dtype=h5py.special_dtype(vlen=dtype), chunks=chunks, **kwds
    124 )
    125 return dset

AttributeError: 'list' object has no attribute 'dtype'
```
## Python environment:

- HyperSpy version: 2.0.0
- rsciio version: 0.3

## Additional context

This does not occur if there are no navigation dimensions.

ericpre, 2024-01-20:
I had a quick look at this and the issue is that h5py doesn't support for these numpy dtype (for example: https://docs.h5py.org/en/stable/strings.html#what-about-numpy-s-u-type). The way around these is to convert the unsupported type to something supported by h5py.

For example, the following example works for saving without changing rosettasciio code:

```python
import hyperspy.api as hs
import numpy as np

# Create a Signal2D with 1 navigation dimension
rng = np.random.default_rng(0)
data = np.ones((10, 100, 100))
s = hs.signals.Signal2D(data)

s2 = hs.signals.Signal2D(data)

import h5py
dt = h5py.special_dtype(vlen=str) 

offsets = np.empty(s.axes_manager.navigation_shape, dtype=object)
texts = np.empty(s.axes_manager.navigation_shape, dtype=object)

for index in np.ndindex(offsets.shape):
    i = index[0]
    offsets[index] = rng.random((10, 2))[:i+2] * 100
    texts[index] = np.array(['a' * (i+1), 'b', 'c', 'd', 'e', 'f', 'g', 'f', 'h', 'i'][:i+2], dtype=dt)

m2 = hs.plot.markers.Texts(
    offsets=offsets,
    texts=texts,
    sizes=3,
    facecolor="black",
    )

s2.add_marker(m2, permanent=True)
s2.plot()
s2.save("test.hspy", overwrite=True)

s3 = hs.load("test.hspy")
```

Plotting the marker is still not working because h5py read the array as bytestring instead of string and hopefully it would be a trivial workaround. Obviously, we can't used h5py dtype when creating the marker but illustrate the idea, but we should be able to convert to "suitable" type in the hspy plugin.

@CSSFrancis, does this make sense to you?

CSSFrancis, 2024-01-20:
@ericpre that seems perfect! I didn't realize h5py had their own string dtype.

There is the warning about fixed length strings truncating, is that something we have to worry about when casting back and forth?

ericpre, 2024-01-20:
What warning are you referring to? I don't have any.

closed 2024-01-31 by ericpre