hyperspy/rosettasciio #228: M4 tornado Micro-XRF bruker .bcf file import error

Divyadeep00, 2024-02-19:
Does Hyperspy support .BCF file of micro-XRF now? I am getting error File format not supported.

I even Tried with RosettaSciIO bcf import function

```
import rsciio 
rsciio.bruker.file_reader(r'MS-1b_S.bcf', cutoff_at_kV='auto')
```

I am getting error

```
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
Cell In[45], line 1
----> 1 rsciio.bruker.file_reader(r'MS-1b_S.bcf', cutoff_at_kV='auto')

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\rsciio\bruker\_api.py:1314, in file_reader(filename, lazy, select_type, index, downsample, cutoff_at_kV, instrument)
   1312 ext = splitext(filename)[1][1:].lower()
   1313 if ext == "bcf":
-> 1314     to_return = bcf_reader(
   1315         filename=filename,
   1316         lazy=lazy,
   1317         select_type=select_type,
   1318         index=index,
   1319         downsample=downsample,
   1320         cutoff_at_kV=cutoff_at_kV,
   1321         instrument=instrument,
   1322     )
   1323 elif ext == "spx":
   1324     to_return = spx_reader(
   1325         filename=filename,
   1326         lazy=lazy,
   1327     )

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\rsciio\bruker\_api.py:1374, in bcf_reader(filename, lazy, select_type, index, downsample, cutoff_at_kV, instrument)
   1346 """
   1347 Reads a bruker ``.bcf`` file and loads the data into the appropriate class,
   1348 then wraps it into a list of dictionaries typically used by the
   (...)
   1370     Can be either 'TEM' or 'SEM'.
   1371 """
   1373 # objectified bcf file:
-> 1374 obj_bcf = BCF_reader(filename, instrument=instrument)
   1375 if select_type == "image":
   1376     return bcf_images(obj_bcf)

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\rsciio\bruker\_api.py:888, in BCF_reader.__init__(self, filename, instrument)
    887 def __init__(self, filename, instrument=None):
--> 888     SFS_reader.__init__(self, filename)
    889     header_file = self.get_file("EDSDatabase/HeaderData")
    890     self.available_indexes = []

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\rsciio\bruker\_api.py:310, in SFS_reader.__init__(self, filename)
    308 # read the file header
    309 with open(filename, "rb") as fn:
--> 310     a = fn.read(8)
    311     if a != b"AAMVHFSS":
    312         raise TypeError("file '{0}' is not SFS container".format(filename))

OSError: [Errno 22] Invalid argument
```

Apologise if I Posted in wrong forum

ericpre, 2024-02-19:
Yes, this is the right place to ask about this.
It should - at least some bcf XRF file. For reference, support was added in https://github.com/hyperspy/hyperspy/pull/2694.

Could you share a file?

Divyadeep00, 2024-02-19:
yeah sure

https://drive.google.com/file/d/1N5wMRe9XPmwQzKv-WoWnvNXXvm2fze5x/view?usp=sharing

ericpre, 2024-02-19:
This works fine for me. From a quick search on internet, `OSError: [Errno 22] Invalid argument` seems to be related to issues with file (incompatible filename, storage).
Can you try to check that there is not anything special or no issue with the file itselt, for example, it is not corrupted when copying, etc.?

Divyadeep00, 2024-02-19:
I am using this 
```
filename="MS-1b_S.bcf"
rsciio.bruker.file_reader(filename, lazy=False, select_type=None, index=None, downsample=1, cutoff_at_kV=None, instrument=None)
```

getting error now
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[5], line 2
      1 filename="MS-1b_S.bcf"
----> 2 rsciio.bruker.file_reader(filename, lazy=False, select_type=None, index=None, downsample=1, cutoff_at_kV=None, instrument=None)

AttributeError: module 'rsciio' has no attribute 'bruker'
```

Divyadeep00, 2024-02-19:
I would request you to kindly make a code snippet (if working fine for you), so I can read .bcf  file and able to see the instrument running conditions like Acceleration voltage, spot size, distance between spots etc etc. And able to extract maps of different elements. and so on. 

Thanks in Advance

Divyadeep00, 2024-02-19:
for me this is not the output 

My output for above code

`ERROR | Hyperspy | If this file format is supported, please report this error to the HyperSpy developers. (hyperspy.io:579)`

with an error
```
--------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Cell In[6], line 3
      1 import hyperspy.api as hs
----> 3 s = hs.load("MS-1b_S.bcf")
      4 print(s)

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\hyperspy\io.py:517, in load(filenames, signal_type, stack, stack_axis, new_axis_name, lazy, convert_units, escape_square_brackets, stack_metadata, load_original_metadata, show_progressbar, **kwds)
    514         objects.append(signal)
    515 else:
    516     # No stack, so simply we load all signals in all files separately
--> 517     objects = [load_single_file(filename, lazy=lazy, **kwds)
    518                for filename in filenames]
    520 if len(objects) == 1:
    521     objects = objects[0]

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\hyperspy\io.py:517, in <listcomp>(.0)
    514         objects.append(signal)
    515 else:
    516     # No stack, so simply we load all signals in all files separately
--> 517     objects = [load_single_file(filename, lazy=lazy, **kwds)
    518                for filename in filenames]
    520 if len(objects) == 1:
    521     objects = objects[0]

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\hyperspy\io.py:576, in load_single_file(filename, **kwds)
    569     raise ValueError(
    570         "`reader` should be one of None, str, "
    571         "or a custom file reader object"
    572     )
    574 try:
    575     # Try and load the file
--> 576     return load_with_reader(filename=filename, reader=reader, **kwds)
    578 except BaseException:
    579     _logger.error(
    580         "If this file format is supported, please "
    581         "report this error to the HyperSpy developers."
    582     )

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\hyperspy\io.py:610, in load_with_reader(filename, reader, signal_type, convert_units, load_original_metadata, **kwds)
    608 if signal_type is not None:
    609     signal_dict['metadata']["Signal"]['signal_type'] = signal_type
--> 610 signal = dict2signal(signal_dict, lazy=lazy)
    611 signal = _add_file_load_save_metadata('load', signal, reader)
    612 path = _parse_path(filename)

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\hyperspy\io.py:792, in dict2signal(signal_dict, lazy)
    789 elif signal_dimension == -1:
    790     # If not defined, all dimension are categorised as signal
    791     signal_dimension = signal_dict["data"].ndim
--> 792 signal = assign_signal_subclass(signal_dimension=signal_dimension,
    793                                 signal_type=signal_type,
    794                                 dtype=signal_dict['data'].dtype,
    795                                 lazy=lazy)(**signal_dict)
    796 if signal._lazy:
    797     signal._make_lazy()

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\hyperspy\io.py:737, in assign_signal_subclass(dtype, signal_dimension, signal_type, lazy)
    733 # Regardless of the number of signals in the dict we assign one.
    734 # The following should only raise an error if the base classes
    735 # are not correctly registered.
    736 for key, value in signal_dict.items():
[... 63 lines cut ...]
File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\exspy\misc\eels\gosh_gos.py:21
      1 # -*- coding: utf-8 -*-
      2 # Copyright 2007-2023 The exSpy developers
      3 #
   (...)
     16 # You should have received a copy of the GNU General Public License
     17 # along with exSpy. If not, see <https://www.gnu.org/licenses/#GPL>.
     19 import logging
---> 21 import h5py
     22 import numpy as np
     23 import pooch

File ~\miniconda3\envs\xrd_analysis\Lib\site-packages\h5py\__init__.py:25
     19 # --- Library setup -----------------------------------------------------------
     20 
     21 # When importing from the root of the unpacked tarball or git checkout,
     22 # Python sees the "h5py" source directory and tries to load it, which fails.
     23 # We tried working around this by using "package_dir" but that breaks Cython.
     24 try:
---> 25     from . import _errors
     26 except ImportError:
     27     import os.path as _op

ImportError: DLL load failed while importing _errors: The specified procedure could not be found.
```

ericpre, 2024-02-19:
Your installation of h5py seems to be broken because based on the traceback error it fails when importing h5py.

Divyadeep00, 2024-02-19:
how to fix this?

ericpre, 2024-02-19:
If you have an issue with your h5py installation, this is not a good place to ask! 
Anyway, reinstalling `h5py` may work...

Divyadeep00, 2024-02-19:
Thanks for your suggestion i am reinstalling it. Thanks a lot

Divyadeep00, 2024-02-19:
Hi now i am able to reinstall H5py and able to import .bcf file but rest function of EDS is not working with .bcf file. all functions are disable for this file

[image]

ericpre, 2024-02-19:
This is normal: it returns a list of hyperspy "signals" and the method you are trying to use only work on signal (not list of signal). Closing as this is working as expected.

Regarding how to process the data, you will need to refer to the relevant documentation, here is a list of user guide and tutorial that should be useful:
- https://hyperspy.org/hyperspy-doc/current/
- https://hyperspy.org/exspy
- https://github.com/hyperspy/hyperspy-demos
- https://github.com/hyperspy/exspy-demos

Divyadeep00, 2024-02-20:
I am facing issues when i am trying to see the X-ray lines on my EDS spectrum
`p.sum().plot(xray_lines=True)`

getting response that my Kalpha lines are not in range, although my data is up to 37 Kev

[image]