hyperspy/rosettasciio #241: Rechunking large BCF files

AbaPanu, 2024-03-23:
Hello!

I work with µXRF BCF files that are around 35 GB large and therefore need to load them in "Lazy" mode. I discovered that hyperspy automatically only loads the files as one dask chunk, which is identical to the 3D dimensions of the "non-Lazy" numpy array (tested with smaller files that can be loaded non-lazy). As a result, I cannot do anything with files that are too large for non-lazy mode, because hyperspy crashes as soon as I .compute(). According to the documentation there are options to "rechunk" or "optimize" the chunks, but no hints on how to use them. Is it possible to rechunk during loading? If not, is it even possible to load large BCF files?

Thanks a lot!

ericpre, 2024-03-23:
@AbaPanu, it is important to keep issue focused on what they are about, otherwise, when trying to understand them, it is confusing and not efficient, e.g. have to read about thing that are irrelevant to the issue at hand.

We will keep this thread focused on "lazy processing of large BCF file".

sem-geologist, 2024-12-02:
I started looking into this, and as far I had looked to Dask documentation, it should be possible to divide reading of BCF into smaller chunks for dask array assembly. I see two possibilities to achieve that: implement numpy like indexing for lazy retrieve of data, or use `map_blocks` from dask. It would really be handy to get a copy of such enormous dataset (tens of GB's) so I could experiment with.

ericpre, 2024-12-02:
I haven't used this format for a long time but I could try to dig into some of my old data! This would be ~1GB only but it would still be useful to specify chunks.