hyperspy/rosettasciio #273: EDS map from recent .emd file: Element selection missing from the metadata

OliDG, 2024-06-11:
#### Describe the bug
Dear developpers,
 
the EDS element selection is visible in the original metadata but not loaded to the metadata.Sample information. (Former .emd format seems to load the list of element correctly, exemples below)

#### To Reproduce
Steps to reproduce the behavior:
```
s = hs.load(file)
s
Out[194]: 
[<EDSTEMSpectrum, title: EDS, dimensions: (|4096)>,
 <EDSTEMSpectrum, title: EDS, dimensions: (|4096)>,
   [...]
 <Signal2D, title: Ti, dimensions: (|308, 308)>,
 <EDSTEMSpectrum, title: EDS, dimensions: (308, 308|4096)>]
```
```
s[-1].original_metadata
EDSSpectrumQuantificationSettings
│   │   ├── 0e3648a4c07a47fd9616289dbe0036f6
│   │   │   ├── absorptionCorrection
│   │   │   ├── backgroundCorrection = /SharedProperties/BackgroundCorrection/107c75affaf84aac9a1edbf2f615a3a9
│   │   │   ├── backgroundWindows = /SharedProperties/BackgroundWindows/e7a2a4876cbd4ddd87628410bef92f51
│   │   │   ├── elementProperties <list>
│   │   │   ├── elementSelection = [22, 23, 24, 29, 31, 41]     <------------
│   │   │   ├── ionizationCrossSectionModel = BrownPowell
```
```
s[-1].metadata
Out[193]: 
├── Acquisition_instrument
[...]
├── General
│   ├── FileIO
│   │   └── 0
│   │       ├── hyperspy_version = 2.0.1
│   │       ├── io_plugin = rsciio.emd
│   │       ├── operation = load
│   │       └── timestamp = 2024-06-11T11:55:48.578441+02:00
│   ├── date = 2024-05-28
│   ├── original_filename = 1401_KEM-H1_SC-Nb16-1_Ex-situ_SI_Diffraction_Nano_776_nm_130_kx.emd
│   ├── time = 14:01:12
│   ├── time_zone = Västeuropa, sommartid
│   └── title = EDS
└── Signal
    └── signal_type = EDS_TEM
```
#### Expected behavior
List of elements loaded under "s[-1].metadata.Sample.elements"
Here is the same using an older .emd file.
```
Acquisition_instrument
[...]
├── General
│   ├── FileIO
│   │   └── 0
│   │       ├── hyperspy_version = 2.0.1
│   │       ├── io_plugin = rsciio.emd
│   │       ├── operation = load
│   │       └── timestamp = 2024-06-11T12:12:36.134231+02:00
│   ├── date = 2023-08-08
│   ├── original_filename = 1041 TMV-C8_posC_ SI Diffraction Nano HAADF-DF4-DF2-BF 550 nm.emd
│   ├── time = 10:41:02
│   ├── time_zone = Västeuropa, sommartid
│   └── title = EDS
├── Sample                                                                  <------------
│   └── elements = ['N', 'O', 'Al', 'Si', 'Cu', 'Zr']                                  <------------
└── Signal
    └── signal_type = EDS_TEM
```
#### Python environement:
 - Velox 3.12.0.1298
 - RosettaSciIO version: 0.4
 - Python version: 3.11.7
 - HyperSpy version: 2.0.1

#### Additional context
Add any other context about the problem here. If necessary, add screenshots to help explain your problem.

ericpre, 2024-06-11:
Is it from a spectrum, a spectrum image or both? Do you have a file to share?

jlaehne, 2024-06-15:
@OliDG it should be fixed in the development branch now.

closed 2024-06-15 by jlaehne

OliDG, 2024-06-18:
Thanks, I have tested it and it mostly works. There is an error generated (maybe here previously?) when no elements were selected for quantification before saving the .emd file from Velox ("elementSelection" key doesn't exist then). 

```
ERROR | Hyperspy | If this file format is supported, please report this error to the HyperSpy developers. (hyperspy.io:600)
caught <class 'AttributeError'>: 'DictionaryTreeBrowser' object has no attribute 'elementSelection'
```
\Olivier

PR #274 (2024-06-15): Fix parsing elements from EDS data from velox emd file v11
Fix #273.
xref #226.

### Progress of the PR
- [x] Fix parsing elements from EDS data from velox emd file v11,
- [n/a] update docstring (if appropriate),
- [n/a] update user guide (if appropriate),
- [x] add a changelog entry in the `upcoming_changes` folder (see [`upcoming_changes/README.rst`](https://github.com/hyperspy/rosettasciio/blob/main/upcoming_changes/README.rst)),
- [x] Check formatting of the changelog entry (and eventual user guide changes) in the `docs/readthedocs.org:rosettasciio` build of this PR (link in github checks) [... cut]