hyperspy/rosettasciio #279: regarding fix #273 - new error raise - "EDS map from recent .emd file: Element selection missing from the metadata"

OliDG, 2024-06-18:
#### Describe the bug
I commented on the closed case but realized it may stay unseen.
@ericpre @jlaehne 
Thanks for the fix, I have tested it and it mostly works. There is an error generated (maybe there previously?) when no elements were selected for quantification before saving the .emd file from Velox ("elementSelection" key doesn't exist then). 

```
ERROR | Hyperspy | If this file format is supported, please report this error to the HyperSpy developers. (hyperspy.io:600)
caught <class 'AttributeError'>: 'DictionaryTreeBrowser' object has no attribute 'elementSelection'
```
\Olivier

ericpre, 2024-06-18:
From reading the recent code changes, it is unlikely that this is a regression. Can you provide a file to reproduce this issue so that it can be added to the test suite? Smallest file possible please?
See https://hyperspy.org/rosettasciio/contributing.html#making-test-data-files for more information on making test file.

OliDG, 2024-06-19:
[SI_empty selection.zip](https://github.com/user-attachments/files/15898142/SI_empty.selection.zip)
Thanks for the reference! here is a test file just for you :)

\Olivier