hyperspy/rosettasciio #290: Problem with sum spectra for bruker M4 tonado .BCF file

Divyadeep00, 2024-07-03:
```
import hyperspy.api as hs

import scipy
# Load your data (replace with actual data loading code)
data = hs.load('MS-1b_S.bcf')
data= data[4]
data.plot()
roi = hs.roi.RectangularROI(left=110, right=125, top=55, bottom=65)

s_roi = roi.interactive(data, recompute_out_event=None, color='C1')
s_roi = roi(data)
roi_sum = hs.interactive(s_roi.sum, recompute_out_event=None)
roi_sum.plot()
```
I am using above code to view the spectrum at my roi, while doing so, I encounter a flat XRF spectra shown in image 
[image]

both orange and red are nearly (even when they overlap) the default eds signal navigator gives a spectrum but the sum plot is giving a flat signal, is it a but or I am doing something wrong.

ericpre, 2024-07-03:
By any chance, do you have any NaN in your data? If so, you should use https://hyperspy.org/hyperspy-doc/dev/reference/api.signals/BaseSignal.html#hyperspy.api.signals.BaseSignal.nansum

Divyadeep00, 2024-07-03:
but if the reason is the NaN value than default view eds viewer should also show flat, but it is showing signals

ericpre, 2024-07-03:
No, the default pointer doesn't take the sum.

Divyadeep00, 2024-07-03:
but the provided image both have same pixel, size, default pointer is giving values but sum pointer not, 
 as you can see in the provided image also.

Divyadeep00, 2024-07-03:
[image]
in this image both pixcel is overlap

ericpre, 2024-07-04:
Have you tried to use `nansum` instead of `sum` as suggested above? e.g. try to adapt your example above to use `roi_sum = hs.interactive(s_roi.nansum, recompute_out_event=None)`.

ericpre, 2024-12-02:
@Divyadeep00, have you tried if it works using `nansum` instead of `sum`?

Divyadeep00, 2024-12-02:
Thank you for your comment, @ericpre. I previously tried the method you suggested, but it didn’t work at that time.

Now, I’m unable to attempt it again because the .bcf file is not loading Properly and not detecting any instrument.

Divyadeep00, 2024-12-02:
Yeah, Sure, please give me some time. I will share it with you. 
Thanks for you support

Divyadeep00, 2024-12-03:
I apology for delay, 

I am getting this error

> 'WARNING | Hyperspy | `signal_type='EDS_SEM'` not understood. See `hs.print_known_signal_types()` for a list of installed signal types or https://github.com/hyperspy/hyperspy-extensions-list for the list of all hyperspy extensions providing signals. (hyperspy.io:744)'

Using this code 

```
%matplotlib qt

import hyperspy.api as hs

import scipy
bcffile=sample
print(f"loading {bcffile} file")
# Load your data (replace with actual data loading code)
data = hs.load(bcffile)

print(f"File Loaded")
data= data[4]
# data.set_signal_type("EDS_SEM")
data.metadata.Acquisition_instrument.SEM.beam_energy = 50
# x1, y1, x2, y2 = latest_roi_coords  # Use the provided list

data.axes_manager[0].scale=1
data.axes_manager[0].index=0
data.axes_manager[1].scale=1
data.axes_manager[1].index=0
data.axes_manager[2].scale=0.01
data.axes_manager[2].index=0

data.plot()
# Create a rectangular ROI
roi = hs.roi.RectangularROI(left=x1, right=x2, top=y1, bottom=y2)

# Apply the ROI to the signal
im_roi = roi.interactive(data, color="red")
s_roi = roi(data)

# Calculate the sum spectrum
sum_spectrum = s_roi.sum(axis=(0, 1))

# Plot the sum spectrum
sum_spectrum.plot(xray_lines=True, xlim=(0,20))
```

File link can be found [here](https://drive.google.com/file/d/1N5wMRe9XPmwQzKv-WoWnvNXXvm2fze5x/view?usp=sharing)

I am using version 2.2.0

`hs.print_known_signal_types()`

this results
[image]

ericpre, 2024-12-03:
This is not an error, but a warning telling you that the data couldn't be loaded as "EDS_SEM" signal type, because the required extension is not installed (in this case the library [exspy](https://hyperspy.org/exspy/)). If you want to know more about possible signal type and extension, have a read at https://hyperspy.org and https://github.com/hyperspy/hyperspy-extensions-list.

If you want something that works out of the box with all extensions installed, we provide a python distribution installer: https://hyperspy.org/hyperspy-bundle/

The interactive sum spectrum works fine with:

```python
%matplotlib qt
import hyperspy.api as hs

s_ = hs.load("MS-1b_S.bcf")
s = s_[-1]
s.plot()

roi = hs.roi.RectangularROI()
s_roi = roi.interactive(s, recompute_out_event=None, color='C1')
roi_sum = hs.interactive(s_roi.sum, recompute_out_event=None)
roi_sum.plot()
```

Please re-open, if this doesn't work for you.

closed 2024-12-03 by ericpre