hyperspy/rosettasciio #296: emd reader Velox ModuleNotFoundError

dbdurham, 2024-07-16:
#### Describe the bug
Tried to read in a spectrum image Velox emd file (EDS) and got: 

ModuleNotFoundError                       Traceback (most recent call last)
Cell In[4], line 1
----> 1 d = hs.load('example.emd')

File ~\.conda\envs\hyperspy\Lib\site-packages\hyperspy\io.py:537, in load(filenames, signal_type, stack, stack_axis, new_axis_name, lazy, convert_units, escape_square_brackets, stack_metadata, load_original_metadata, show_progressbar, **kwds)
    533         objects.append(signal)
    534 else:
    535     # No stack, so simply we load all signals in all files separately
    536     objects = [
--> 537         load_single_file(filename, lazy=lazy, **kwds) for filename in filenames
    538     ]
    540 if len(objects) == 1:
    541     objects = objects[0]

File ~\.conda\envs\hyperspy\Lib\site-packages\hyperspy\io.py:596, in load_single_file(filename, **kwds)
    590     raise ValueError(
    591         "`reader` should be one of None, str, " "or a custom file reader object"
    592     )
    594 try:
    595     # Try and load the file
--> 596     return load_with_reader(filename=filename, reader=reader, **kwds)
    598 except BaseException:
    599     _logger.error(
    600         "If this file format is supported, please "
    601         "report this error to the RosettaSciIO developers at "
    602         "https://github.com/hyperspy/rosettasciio/issues"
    603     )

File ~\.conda\envs\hyperspy\Lib\site-packages\hyperspy\io.py:618, in load_with_reader(filename, reader, signal_type, convert_units, load_original_metadata, **kwds)
    616 lazy = kwds.get("lazy", False)
    617 if isinstance(reader, dict):
--> 618     file_data_list = importlib.import_module(reader["api"]).file_reader(
    619         filename, **kwds
    620     )
    621 else:
    622     # We assume it is a module
    623     file_data_list = reader.file_reader(filename, **kwds)

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\emd\_api.py:201, in file_reader(filename, lazy, dataset_path, stack_group, select_type, first_frame, last_frame, sum_frames, sum_EDS_detectors, rebin_energy, SI_dtype, load_SI_image_stack)
    199         raise IOError("The file is not a supported EMD file.")
    200 except Exception as e:
--> 201     raise e
    202 finally:
    203     if not lazy:

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\emd\_api.py:189, in file_reader(filename, lazy, dataset_path, stack_group, select_type, first_frame, last_frame, sum_frames, sum_EDS_detectors, rebin_energy, SI_dtype, load_SI_image_stack)
    177     _logger.debug("EMD file is a Velox variant.")
    178     emd_reader = FeiEMDReader(
    179         lazy=lazy,
    180         select_type=select_type,
   (...)
    187         load_SI_image_stack=load_SI_image_stack,
    188     )
--> 189     emd_reader.read_file(file)
    190 elif is_EMD_NCEM(file):
    191     from ._emd_ncem import EMD_NCEM

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\emd\_emd_velox.py:136, in FeiEMDReader.read_file(self, f)
    134 if self.im_type == "SpectrumStream":
    135     self._parse_image_display(f)
--> 136 self._read_data(self.select_type)

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\emd\_emd_velox.py:166, in FeiEMDReader._read_data(self, select_type)
    164 self._read_images()
    165 t1 = time.time()
--> 166 self._read_spectrum_stream()
    167 t2 = time.time()
    168 _logger.info("Time to load images: {} s.".format(t1 - t0))

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\emd\_emd_velox.py:584, in FeiEMDReader._read_spectrum_stream(self)
    582     _logger.warning("The file contains only one spectrum stream")
    583 # Read the first stream
--> 584 s0 = _read_stream(subgroup_keys[0])
    585 streams = [s0]
    586 # add other stream streams

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\emd\_emd_velox.py:577, in FeiEMDReader._read_spectrum_stream.<locals>._read_stream(key)
    576 def _read_stream(key):
--> 577     stream = FeiSpectrumStream(spectrum_stream_group[key], self)
    578     return stream

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\emd\_emd_velox.py:931, in FeiSpectrumStream.__init__(self, stream_group, reader)
    929     self.spectrum_image = self.stream_to_sparse_array(stream_data=stream_data)
    930 else:
--> 931     self.spectrum_image = self.stream_to_array(stream_data=stream_data)

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\emd\_emd_velox.py:975, in FeiSpectrumStream.stream_to_array(self, stream_data, spectrum_image)
    964 def stream_to_array(self, stream_data, spectrum_image=None):
    965     """Convert stream to array.
    966 
    967     Parameters
   (...)
    973 
    974     """
--> 975     import rsciio.utils.fei_stream_readers as stream_readers
    977     spectrum_image = stream_readers.stream_to_array(
    978         stream=stream_data,
    979         spatial_shape=self.reader.spatial_shape,
   (...)
    986         dtype=self.reader.SI_data_dtype,
    987     )
    988     return spectrum_image

File ~\.conda\envs\hyperspy\Lib\site-packages\rsciio\utils\fei_stream_readers.py:21
     19 import dask.array as da
     20 import numpy as np
---> 21 import sparse
     23 from rsciio.utils.tools import jit_ifnumba
     26 class DenseSliceCOO(sparse.COO):

ModuleNotFoundError: No module named 'sparse'

#### To Reproduce
import hyperspy.api as hs
d = hs.load('example.emd')

#### Expected behavior
Should load the dataset

#### Python environment:
 - RosettaSciIO version: 0.6
 - Python version: 3.12.3
 - HyperSpy version: 2.1.1. (if relevant)
 - HyperSpy extensions and version used

dbdurham, 2024-07-16:
Just realized it is a dependency issue on my end, closing this.

ericpre, 2024-07-17:
Reopening this issue because it should fail with a better error message saying that the `sparse` library is required to read EDS spectrum image from Velox emd file.

Marking this issue as first good issue because it simply need to catch the error at the right place with a try, except block and raise an error with a more suitable message.

mattzitunes, 2024-07-22:
I am getting the same problem as yours. How did you solve it?

ericpre, 2024-07-22:
You need to install the optional [`sparse`](https://sparse.pydata.org/en/stable/index.html) library.
This is mentioned in the [documention](https://hyperspy.org/rosettasciio/user_guide/supported_formats/emd.html#emd-velox) but the error message should be more clear.

ericpre, 2025-06-28:
Done in #305.