hyperspy/rosettasciio #301: Renishaw reader calculates the scale for the microscope image incorrectly

Asharits, 2024-08-13:
#### Describe the bug

In the Renishaw reader, the scale for the microscope image (WHTL) is being calculated incorrectly for hyperspectral maps. This causes the mapped area (red box) to show up in the wrong area. 

#### To Reproduce

I verified this by comparing images to mapped area in the WiRE versus what Hyperspy is plotting for multiple datasets.

#### Expected behavior

I am not sure where in the reader this is being calculated but it is an easy fix. In the original metadata for the microscope image, FocalPlaneXYOrigins contains the origin for X and Y. FocalPlaneXResolution and FocalPlaneYResolution contains the length of the image. These values just need to be added to the origin values and then divided by the image shape to obtain the correct scale.

I tested this out on some of my datasets and it fixed the scale and the red box was now in the correct place.

#### Python environment:
 - RosettaSciIO version: 0.6
 - Python version: 3.11.8
 - HyperSpy version: 2.1.1

ericpre, 2024-08-13:
Are you saying that the offset and scale of the microscope image are incorrect but the hyperspectral maps are correct?

The scale and offset of the microscope image are calculated in https://github.com/hyperspy/rosettasciio/blob/e49911047c03465a84facac546053576a23ef915/rsciio/utils/image.py#L42-L70

And what it does seems to be consistent with your description of how it should be calculated.

As I remember, the data that I have doesn't have this issue. Can you share files and screenshots?

Asharits, 2024-08-14:
Hmm, I wonder why it isn't working for me. Below is a screenshot of the same dataset from WiRE and in Hyperspy. You can see the map are does not match up.

WiRE Screenshot
[image]

Hyperspy Screenshot
[image]

Asharits, 2024-08-14:
To me it looks like an issue with the way scale is calculated using FieldOfViewXY. Maybe there is a difference depending on the version of WiRE? I am on 5.6

Asharits, 2024-10-08:
@ericpre Looking back at this I misspoke. You only need to change fields_of_views to be equal to (FocalPlaneXResolution, FocalPlaneYResolution). Doing this yields the correct map area for me.

[image]

ericpre, 2024-10-19:
@Asharits, can you share a file?

Asharits, 2024-10-24:
Yes, here is a file: [532nm_1800 map.zip](https://github.com/user-attachments/files/17510847/532nm_1800.map.zip)

Here is the screenshot of the file from WiRe: 
[image]

Here is what Hyperspy plots before the fix:
[image]

Here is it after:
[image]

ericpre, 2024-10-24:
@Asharits, thank you for sharing the file.
Your suggested changes seems to be correct. The data that I used as reference to parse the calibration, have been acquired with WiRE 5.2. The data saved with the two different version of software (WiRE 5,2 and 5.6) doesn't have the same metadata: in case of WiRE 5.2 `FieldOfViewXY` are the same values as (`FocalPlaneXResolution`, `FocalPlaneYResolution`) and therefore doesn't have the issue. For data saved with WiRE 5.6, these values are different!

I will do some check when I have a chance but it all looks consistent. I made a PR with the fix: #327.

PR #327 (2024-10-25): Fix scale in white field image in renishaw reader
Fix #301.

### Progress of the PR
- [x] Fix scale in white field image in renishaw reader,
- [n/a] update docstring (if appropriate),
- [n/a] update user guide (if appropriate),
- [x] add a changelog entry in the `upcoming_changes` folder (see [`upcoming_changes/README.rst`](https://github.com/hyperspy/rosettasciio/blob/main/upcoming_changes/README.rst)),
- [x] Check formatting of the changelog entry (and eventual user guide changes) in the `docs/readthedocs.org:rosettasciio` build of this PR (link in github checks)
- [x] update tests,
- [x] re [... cut]
closed 2024-10-25 by jlaehne