hyperspy/rosettasciio #312: QT Based File Translation Application

CSSFrancis, 2024-09-16:
#### Describe the functionality you would like to see.
I think there is a use for a basic qt based application which can translate files easily and works for batch conversion of file types. This might be useful for people who are more comfortable using GUI's or when working on a microscope. 

Sometimes it is also easier just to Queue up a lot of files when you are trying to batch convert.

I made a simple application that does something like this:

[image]

I need to handle errors better and maybe add some more features but it's a pretty lightweight and useful application I think.

CSSFrancis, 2024-09-16:
> Indeed, many colleagues will appreciate such a batch converter.

That's been my impression from talking to people lately. 

> And it can be a nice advertisement for the HyperSpy project (e.g. open an About window when clicking on the Logo).

Good idea!

ericpre, 2024-09-16:
That would be very good. I can come up with a list of useful features, but these can be added over time! ;)

I have been using something like that for many years but it is quite rough (and very specific to the way I handle my data and some of my need at the time...): https://github.com/ericpre/BatchOperationTool.