hyperspy/rosettasciio #314: Reliscensing .mrc file Loader

CSSFrancis, 2024-09-23:
Maybe it is about time to think about reliscencing some code bit by bit. 

LiberTEM is planning on re-licensing to a more premissive license: https://github.com/LiberTEM/LiberTEM/issues/1649 

Currently some of the .mrc file loading is [broken](https://github.com/LiberTEM/LiberTEM/issues/1675) in LiberTEM and it might be nice to just have a single file loader to support rather than supporting multiple different versions of the same thing.

#### Describe the functionality you would like to see.

@sk1p Maybe you can better explain what license you would need to refactor? But my understanding is that LGPL should be sufficient?  Otherwise we could maybe consider MIT?

#### Describe the context

Just pining the people who have contributed to the .mrc file loader:

@ericpre @jlaehne @francisco-dlp

CSSFrancis, 2024-09-23:
Also @pietsjoh 😅

sk1p, 2024-09-23:
See also #51 for the general question

ericpre, 2024-09-25:
The "bit by bit" re-licencing may be messy in term of distribution (make separate package) and getting approval to change from all contributors will take some time.

@CSSFrancis, if you think that it would be beneficial to change the license of the MRC reader in a shorter timeframe than re-licensing the whole of rosettasciio, then one option would be to include the MRC code where it is needed. Maybe this could be done by as simple copy and paste while keeping the rosettasciio as upstream or doing some git wizardery involving submodule and filter-branch...

In term of license, my view is that a more permissive license will be the simplest because it seems that there are ambiguities around the use of LGPL.

jlaehne, 2024-09-25:
I would not object to a reliscensing or either a part or the full package. The biggest hurdle seemed to clarify what the best choice of license is, but as far as I remember a number of points were indeed pointing towards MIT being the best choice.

It might though be hard to get explicit consent from all contributors as some might not be active on github any longer. I currently have no idea what the best way forward would be on that.