hyperspy/rosettasciio #316: Failed assert for lazily-loaded tiff stack

realitychemist, 2024-10-03:
#### Describe the bug
I have simulated 4D-STEM data (output from the Dr Probe software) that's stored as a 3D TIFF stack with a companion JSON file that contains the metadata (e.g. the correct shape for the 4D axes). I am trying to load the TIFF file, reshape it according to the information in the JSON file, and save the reshaped data in EMD (NCEM) format (which is the native file format for py4DSTEM).

The 3D TIFF stack is ~180 GB in size and will not fit in RAM, so I am loading it lazily via `tiff.file_reader(tiff_fname, lazy=True)`. This appears to work at first: I can (lazily) create a reshaped `dask` array based on the JSON file (loaded separately with the `json` library) and I can construct the `axes` and `metadata` objects with no issue (all necessary information is in the JSON file).

When I go to actually write out the new EMD file (via `emd.file_writer(emd_fname, signal)`) an `AssertionError: assert self._fh is not None` is raised. Here's the stack trace (with paths trimmed up to "site-packages" for readability):

> Traceback (most recent call last):
  File "\<ipython-input-7-6d09027a0035\>", line 2, in <module>
    emd.file_writer(emd_fname, signal)
  File "...\\rsciio\emd\_api.py", line 230, in file_writer
    EMD_NCEM().write_file(filename, signal, chunks=chunks, **kwds)
  File "...\\rsciio\emd\_emd_ncem.py", line 474, in write_file
    self._write_signal_to_group(signal_group, signal, **kwargs)
  File "...\\rsciio\emd\_emd_ncem.py", line 500, in _write_signal_to_group
    dataset.create_dataset(
  File "...\\h5py\_hl\group.py", line 183, in create_dataset
    dsid = dataset.make_new_dset(group, shape, dtype, data, name, **kwds)
  File "...\\h5py\_hl\dataset.py", line 46, in make_new_dset
    data = array_for_new_object(data, specified_dtype=dtype)
  File "...\\h5py\_hl\base.py", line 118, in array_for_new_object
    data = np.asarray(data, order="C", dtype=as_dtype)
  File "...\\dask\array\core.py", line 1693, in __array__
    x = self.compute()
  File "...\\dask\base.py", line 375, in compute
    (result,) = compute(self, traverse=False, **kwargs)
  File "...\\dask\base.py", line 661, in compute
    results = schedule(dsk, keys, **kwargs)
  File "...\\rsciio\tiff\_api.py", line 399, in _load_data
    dc = handle.asarray(out=memmap)
  File "...\\tifffile\tifffile.py", line 12357, in asarray
    result = self.parent.asarray(series=self, **kwargs)
  File "...\\tifffile\tifffile.py", line 4492, in asarray
    result = self.filehandle.read_array(
  File "...\\tifffile\tifffile.py", line 14842, in read_array
    assert self._fh is not None
AssertionError

I might be off base, but it looks to me like maybe the original TIFF file is not being kept open through the duration of the script? Hence `tifffile`'s complaint about there not being any filehandle.  I did try to do this with the `with open(tiff_fname, "r") as f:` pattern to have more control over when the file is closed, but as far as I can tell `tiff.file_reader` can't accept a file-like object, only a string or path-like object.

#### To Reproduce
This seems to happen with any size of TIFF stack (I only have a couple to test with at the moment, but it happens with the 180 GB file as well as a little 40 MB file I've got) which should mean it isn't too hard to reproduce. In practice there's not much reason to do this lazily unless the file is too big to fit in memory (as mine is) and there's no error in the eager case.

Here are the relevant parts of the code (metadata handling omitted for brevity), it's fairly straightforward:
```py
import json
from rsciio import tiff, emd

meta_fname = "path\to\json.json"
with open(meta_fname, "r") as f:
    input_meta = json.load(f)

tiff_fname = "path\to\tiff.tif"
input_tiff = tiff.file_reader(tiff_fname, lazy=True)
raw_data = input_tiff[0]["data"]

shaped_data = raw_data.reshape((input_meta["scan"]["dims"][0],
                                input_meta["scan"]["dims"][1],
                                input_meta["diffraction"]["dims"][0],
                                input_meta["diffraction"]["dims"][1]))

# ... then some code that translates metadata from the json file to the rsciio dictionary structure ...

signal = {"data": shaped_data, "axes": axes, "metadata": metadata, "original_metadata": {}}
emd_fname = "path\to\save\emd.emd"
emd.file_writer(emd_fname, signal)
```

I found that I could reproduce this with other operations that force dask to switch from lazy to eager, such as simply calling `shaped_data.compute()`, but the above is closer to my actual code.

#### Expected behavior
I expected `dask` to handle everything lazily right up until `emd.file_writer` was called, and then to begin going through its task graph to actually read the data out of the TIFF file and write it back to the EMD file in the new shape. I am not sure exactly what the task graph looks like in this case, but if that might be useful I can install `graphviz` and check. (This is a shared workstation, installs which require a reboot - like `graphviz` - take a bit of coordination.)

#### Python environement:
 - RosettaSciIO version: 0.6
 - Python version: 3.10.9
 - HyperSpy version: 2.1.0
 - Dask version: 2024.5.2
 - tifffile version: 2024.5.22

ericpre, 2024-10-04:
To understand the issue, it would be useful to have a minimal reproducible example. Can you make an example that also create the image, save it as tif, and then do the conversion from tiff to emd file. Maybe with a single image and without the reshaping, axes, etc. The image can be created from numpy, doesn't need to be a real image at this stage.

realitychemist, 2024-10-04:
Sure thing:
```py
from rsciio import tiff, emd
import numpy as np

dummy_data = np.random.random_sample(size=(10, 256, 256))
z = {"name": "depth", "index_in_array": 0, "navigate": True,
     "size": 10, "scale": 1, "offset": 0, "units": ""}
x = {"name": "scanx", "index_in_array": 1, "navigate": False,
     "size": 256, "scale": 1, "offset": 0, "units": ""}
y = {"name": "scany", "index_in_array": 2, "navigate": False,
     "size": 256, "scale": 1, "offset": 0, "units": ""}
tiff.file_writer("dummy.tiff", {"data": dummy_data, "axes": [z, x, y],
                                "metadata": {"General": {"title": "dummy"}, "Signal": {}}})
from_tiff = tiff.file_reader("dummy.tiff", lazy=True)
emd.file_writer("test.emd", {"data": from_tiff[0]["data"], "axes": [z, x, y],
                             "metadata": {"General": {"title": "test"}, "Signal": {}}, "original_metadata": {}})
```
Here I'm writing the dummy data out to a `.tiff`, then immediately reading that data back and trying to write it to `.emd`. This raises the same `AssetionError`, at least on my machine.

ericpre, 2024-10-05:
@realitychemist: this should be fixed in #317.

closed 2024-10-19 by ericpre