hyperspy/rosettasciio #323: Issue with loading Bruker .bcf and .spx files

brtkok, 2024-10-17:
I was trying to use Hyperspy for data analysis as I prefer to do all that stuff in Python. Unfortunately, I'm getting the below error when trying to just load the file. I tried many different things, using different formats (.bcf and .spx) using the full file path of the file, trying the 'lazy' loading, but I seem to be stuck.

There's not much code to share, since it's literally one of the first steps, but this is what I did:

`import hyperspy.api as hs`
`spectrum = hs.load('JCBbattery30x.bcf')`

What I'm getting is `hyperspy.io:599` and then `KeyError: 'PrimaryEnergy'`. I'm just wondering what could be the source of this, did I save the file incorrectly or is it the module issue? It works just fine in Bruker's Quantax70...

Python environement:
 - Python version: 3.12.4
 - HyperSpy version: 2.1.1

CSSFrancis, 2024-10-17:
@ihatechickpeas Out of curioisity what type of data are you trying to load?

Here is some more infromation: 

https://hyperspy.org/rosettasciio/user_guide/supported_formats/bruker.html#bruker-format

It's likely that the change required isn't terribly large to support your data.  Do you have a smallish data set that we could potentially use for testing?  Often times the biggest hurdle to reading is having the right dataset to compare against.

ihatechickpeas, 2024-10-22:
@CSSFrancis it's a battery cross section scan that contains the sample scan with EDX data of it. I've attached the data file here:

 [JCBbattery30x.zip](https://github.com/user-attachments/files/17474470/JCBbattery30x.zip)

I talked to a colleague of mine and he did say it's possible to override the Primary Energy error with RosettaSciIO module by adding the energy manually, but I haven't tried that yet. Still, if it's a common problem with Bruker format, or just data sets from this specific type of microscopes we're using, I think it'd be great if you managed to figure out what the issue is exactly and if it's possible to easily fix it :)

Thanks a lot for the help!

ericpre, 2024-10-22:
@ihatechickpeas, thank you for sharing the file. Would you able to provide a smaller file (smallest possible)? If so, please have a look at our [documentation](https://hyperspy.org/rosettasciio/contributing.html#making-test-data-files) for explanation and recommendation.
https://github.com/hyperspy/rosettasciio/pull/326 should fix the issue and it needs small test file.

sem-geologist, 2024-11-25:
@ihatechickpeas, AFAIK, Bruker Esprit allows to select 16x16 ROI as smallest  area for Hyperspectra. That way the smallest file example can be made.

proposed workaround is short-term fix, as Hyperspy (AFAIK, please correct me) has no XRF signal. Hopefully more universal metadata structure will be setup by community, and XRF then can get full support. Additional obstacly you probably will get into is wrong live time support in Hyperspy (it expects a scalar instead of array (raster of livetimes corresponding to every pixel)). It is not a severe problem for material scientists, but absolute stopper for geoscientists, and often huge obstacle for XRF maps.

sem-geologist, 2024-12-02:
The idea for test files:
instead of using huge test files for every testing case (especially where tests only check particular metadata absence in header) lets make an artificial files (with 0s in data cube) with header copied from test file. Or (to not fiddle around with packing/unpacking and creating new bcf files, test could reuse same XRF bcf file which this lib already includes, and overwrite the missing part of XML (of this issue) with white spaces on temporary copy of the test file. So there could be one "the mother of all test files" file from which other test files could be made in-place for testing.

ihatechickpeas, 2024-12-12:
@ericpre @sem-geologist apologies for the delay, I was hoping to get it done this year, but I'm having a slight delay. As I'm still getting trained I can't take the data on my own. I should be able to access the microscope in January / early February, I will update you as soon as I can.

ihatechickpeas, 2025-02-13:
@ericpre I finally managed to access the lab and get a test file. I've tried it out before posting it here, and... It actually worked this time, partially, at least. I loaded the file, so good so far, I figured out it stored two separate datasets in a list form. The second one was the EDX spectrum I needed. I tried to plot the data, but unfortunately, all I got was an (I believe) SE image and the EDX plot in the form of a flat line. I found it weird that the code returned an SE plot... 

About the flat EDX line, here you could argue it's just the type of signal I got, but it does look like a regular spectrum in Bruker software, one that actually contains some information. Additionally, I checked another file with data collected over 1h... Same story.

One thing I didn't see in any online tutorial I looked up is that under the EDX spectrum, I've got two axes (I checked it through axes_manager), one called Navigation Axis and the second called Signal Axis. I tried using as_signal1D(0) as described in the tutorial, but it didn't help at all. It just swaps the order of axes (moving signal to navigation), which I guess makes sense, but isn't particularly helpful. I've attached the test file I used, and here's my code:

`import numpy as np
import matplotlib.pyplot as plt
import hyperspy.api as hs
import rsciio
import exspy

s = hs.load("JCBnew_EDX_40x_1min-test.bcf")
d = s[0]
f = s[1]

f = f.as_signal1D(0)
f.set_signal_type("EDS_SEM")

f.metadata.Acquisition_instrument.SEM.elevation_angle = 22
#f.metadata.Acquisition_instrument.SEM
#f.axes_manager

f.plot()`

[JCBnew_EDX_40x_1min-test.zip](https://github.com/user-attachments/files/18784736/JCBnew_EDX_40x_1min-test.zip)

ericpre, 2025-02-13:
> The second one was the EDX spectrum I needed. I tried to plot the data, but unfortunately, all I got was an (I believe) SE image and the EDX plot in the form of a flat line. I found it weird that the code returned an SE plot...

It looks like a flat, because there is no signal in that navigation coordinate. If you take the sum over all navigation coordinates, you will see that there is a EDS spectrum:

```python
import hyperspy.api as hs

s = hs.load("JCBnew_EDX_40x_1min-test.bcf")
s[1].sum().plot()
```

Will you be able to figure out why this file works fine and not the other one? It must have something to do with how the data has been acquired, maybe a setting somewhere? This would be useful to understand why in some files, the energy is not set but also to generate a test file.

Can you make a smaller test file as discussed above? For example a 16x16 TOI as mentioned in @sem-geologist's comment [above](https://github.com/hyperspy/rosettasciio/issues/323#issuecomment-2498172609)?

ihatechickpeas, 2025-03-11:
@ericpre thanks a lot, it worked! 

I've no idea what's causing the issue, as the procedure for collecting data was exactly the same... I will make sure to always make a test file. Just want to add that the lab I'm using workes with old version of Bruker Quantax, not Esprit. The last test file I created was the smallest area size available. I can however further reduce data collection time (say 10s?).

I am soon getting trained to use another intrument. I'll get a chance to ask more questions about data analysis, so hopefully I'll also gain some more insights. I'll keep posting updates under this thread in case of any more anomalous data files.

ericpre, 2025-03-11:
If this is not possible to reproduce then maybe this was a one off and there is not much to do about it. Also at some point it is trade off between what it is worth and the amount of effort!

In the interest of keeping this thread focused on the original issue (the `KeyError: 'PrimaryEnergy'` error), I would suggest to close this PR once #326 is merged. For any issue, open a different one, otherwise, the discussion get diluted in irrelevant content.

closed 2025-03-22 by ericpre

xmp4660, 2025-03-22:
This is not a random bug; it appears that files exported using Quantax70 all have this issue. Below are these files :

[Mapping.zip](https://github.com/user-attachments/files/19402445/Mapping.zip)

ericpre, 2025-03-22:
@xmp4660, the fix has been merged but no test was included because the file originally provided are too large to be included in the test suite. Are you able to provide small test file as mentioned in https://github.com/hyperspy/rosettasciio/issues/323#issuecomment-2430364313 and https://github.com/hyperspy/rosettasciio/issues/323#issuecomment-2498172609? 

The file that you provided in https://github.com/hyperspy/rosettasciio/issues/323#issuecomment-2745207734 are also too large.

xmp4660, 2025-03-31:
I replaced the old files in the C:\Users\xxx\HyperSpy-bundle\Lib\site-packages\rsciio\bruker directory with the [new files](https://github.com/hyperspy/rosettasciio/tree/main/rsciio/bruker).
But I still need to use `s[1].sum.plot()` to generate the correct plot. Maybe it's my usage method that's the problem.
Additionally, I tried to create the smallest possible bcf file, which is still 2.08MB (can be compressed to 904KB). If needed, I will upload this file.

PR #326 (2025-03-22): Bruker XRF fix
Fix #323.

### Progress of the PR
- [x] Don't raise error when the beam energy can't be found in bcf xrf files,
- [x] Add `export_xml` utility function, 
- [x] update docstring (if appropriate),
- [x] update user guide (if appropriate),
- [x] add a changelog entry in the `upcoming_changes` folder (see [`upcoming_changes/README.rst`](https://github.com/hyperspy/rosettasciio/blob/main/upcoming_changes/README.rst)),
- [x] Check formatting of the changelog entry (and eventual user guide changes) in the `docs/readthedocs.org:rosettasciio` build of this PR (link in github checks)
- [x [... cut]