hyperspy/rosettasciio #333: Velox(.emd)  load gives error in HyperSpy 2

PereraUmesha, 2024-11-12:
#### Describe the bug
EDS Velox (.emd) file that can be loaded and processed using HyperSpy 1.7, gives errors with the same code in HyperSpy.

#### To Reproduce
Steps to reproduce the behavior:
```
import hyperspy.api as hs
test_data = hs.load('Velox_EDS-SI.emd')
print(test_data)

spec_data = hs.load('Velox_EDS-SI.emd', select_type="spectrum_image")
print(spec_data.get_lines_intensity()
```

#### Expected behavior
1) Load the emd file and print the content
2) Load the spectrum_image datea from emd file and print the lines intensity information

#### Python environement:
 - RosettaSciIO version: 0.6
 - Python version: 3.9
 - HyperSpy version: 2.1.1 (if relevant)
 - sparse: 0.15.4
 - exspy: 0.3.1 (The error is different depending on exspy is installed or not)

#### Additional context
When exspy is installed, error occurs at hs.load()

> ERROR | Hyperspy | If this file format is supported, please report this error to the RosettaSciIO developers at https://github.com/hyperspy/rosettasciio/issues (hyperspy.io:599)

When exspy is not installed, error occurs at get_lines_intensity()

> 'Signal1D' object has no attribute 'get_lines_intensity'

ericpre, 2024-11-12:
To use `get_lines_intensity`, you need `exspy` to be installed, because it is a method of the [`EDSSpectrum`](https://exspy.readthedocs.io/en/latest/reference/signals.html#exspy.signals.EDSSpectrum.get_lines_intensity) class.

Can you provide the full error message when loading the file?

PereraUmesha, 2024-11-13:
Please find below the error messages:
(Please note that, not all .emd files give errors I've mentioned. Only .emd files from some EDS equipments give this error with HyperSpy2. All .emd files are working well with HypeerSpy 1.7) 

**In exspy installed environment: [error occurs at hs.load()]**

> 2024/11/12 10:03:12:744:WARNING:`signal_type='EDS_TEM'` not understood. See `hs.print_known_signal_types()` for a list of installed signal types or https://github.com/hyperspy/hyperspy-extensions-list for the list of all hyperspy extensions providing signals.
2024/11/12 10:03:12:197:INFO:emd_spec_data: <Signal1D, title: EDS, dimensions: (512, 512|4096)>
2024/11/12 10:03:13:175:INFO:output data/main_image/Velox_EDS-SI_EDS-spectrum.png
<Signal1D, title: EDS, dimensions: (512, 512|4096)>
2024/11/12 10:03:13:39:ERROR:Traceback (most recent call last):
  File "/home/umesha/hyperspy2_test/20241111_eds_velox_wo_exspy/code/main.py", line 34, in main
    emd2img.main()  
  File "/home/umesha/hyperspy2_test/20241111_eds_velox_wo_exspy/code/modules/emd2img.py", line 198, in main
    get_spec_images(emd_spec_data, emdfile.name[:-4])
  File "/home/umesha/hyperspy2_test/20241111_eds_velox_wo_exspy/code/modules/emd2img.py", line 178, in get_spec_images
    make_element_image(spec, name)
  File "/home/umesha/hyperspy2_test/20241111_eds_velox_wo_exspy/code/modules/emd2img.py", line 96, in make_element_image
    print(signal.get_lines_intensity())
AttributeError: 'Signal1D' object has no attribute 'get_lines_intensity'
Traceback (most recent call last):
  File "/home/umesha/hyperspy2_test/20241111_eds_velox_wo_exspy/code/main.py", line 34, in main
    emd2img.main()  
  File "/home/umesha/hyperspy2_test/20241111_eds_velox_wo_exspy/code/modules/emd2img.py", line 198, in main
    get_spec_images(emd_spec_data, emdfile.name[:-4])
  File "/home/umesha/hyperspy2_test/20241111_eds_velox_wo_exspy/code/modules/emd2img.py", line 178, in get_spec_images
    make_element_image(spec, name)
  File "/home/umesha/hyperspy2_test/20241111_eds_velox_wo_exspy/code/modules/emd2img.py", line 96, in make_element_image
    print(signal.get_lines_intensity())
AttributeError: 'Signal1D' object has no attribute 'get_lines_intensity'

_**pip list**_
> certifi             2024.8.30
> chardet             5.2.0
> charset-normalizer  3.4.0
> click               8.1.7
> cloudpickle         3.1.0
> contourpy           1.3.0
> cycler              0.12.1
> dask                2024.8.0
> flexcache           0.3
> flexparser          0.4
> fonttools           4.54.1
> fsspec              2024.10.0
> h5py                3.12.1
> hyperspy            2.2.0
> idna                3.10
> imageio             2.36.0
> importlib_metadata  8.5.0
> importlib_resources 6.4.5
> Jinja2              3.1.4
> kiwisolver          1.4.7
> lazy_loader         0.4
> llvmlite            0.43.0
> locket              1.0.0
> MarkupSafe          3.0.2
> matplotlib          3.9.2
> mpmath              1.3.0
> natsort             8.4.0
> networkx            3.2.1
> numba               0.60.0
> numpy               2.0.2
> packaging           24.2
> partd               1.4.2
> pillow              11.0.0
> Pint                0.24.4
> pip                 22.0.4
> platformdirs        4.3.6
> pooch               1.8.2
> prettytable         3.12.0
> pyparsing           3.2.0
[... 110 lines cut ...]
> pip                 22.0.4
> platformdirs        4.3.6
> pooch               1.8.2
> prettytable         3.12.0
> pyparsing           3.2.0
> python-box          7.2.0
> python-dateutil     2.9.0.post0
> PyYAML              6.0.2
> requests            2.32.3
> rosettasciio        0.6
> scikit-image        0.24.0
> scipy               1.13.1
> setuptools          58.1.0
> six                 1.16.0
> sparse              0.15.4
> sympy               1.13.3
> tifffile            2024.8.30
> toolz               1.0.0
> tqdm                4.66.6
> traits              6.4.3
> typing_extensions   4.12.2
> urllib3             2.2.3
> wcwidth             0.2.13
> zipp                3.20.2
>

ericpre, 2024-11-14:
Based on the last part of the error message:

> ```python
> import pooch
> File "/home/umesha/hyperspy2_test/20241101_EDS_Velox/lib/python3.9/site-packages/pooch/**init**.py", line 19, in
> from .processors import Unzip, Untar, Decompress
> File "/home/umesha/hyperspy2_test/20241101_EDS_Velox/lib/python3.9/site-packages/pooch/processors.py", line 15, in
> import lzma
> File "/home/umesha/local/lib/python3.9/lzma.py", line 27, in
> from _lzma import *
> ModuleNotFoundError: No module named '_lzma'
> ```

It seems that your python installation is broken. If you have issue with installation, the easiest is to use the hyperspy bundle installer, other methods will need better understanding of how python distribution works.

ericpre, 2024-12-02:
Closing assuming that this is an broken installation of the `lzma` module. @PereraUmesha, please re-open if this is not the case.