hyperspy/rosettasciio #344: Making a New Release

CSSFrancis, 2024-12-11:
@ericpre Any chance we could think about making a new release?  There are some fixes to the .mrc file loader which would be nice to have.

ericpre, 2024-12-11:
Please go ahead, the steps are described in https://github.com/hyperspy/rosettasciio/blob/main/releasing_guide.md.

I think #339 is ready and should included in the next release.

CSSFrancis, 2024-12-11:
Sounds good I will make the relase tomorrow!

CSSFrancis, 2024-12-12:
Outstanding PR's before release:

- [ ]  #324 
- [ ] #339

CSSFrancis, 2024-12-12:
It seems the test suite hadn't been run and currently fails with the newest version of (dask?).  Looking into that now....

ericpre, 2024-12-13:
Release of rosettasciio are quite straightforward (except when CI fails! 😅) and quick to allow releasing as often as necessary, so best to make multiple releases rather than holding a release for a PR.

jlaehne, 2025-02-06:
Addressed by #353

closed 2025-02-06 by jlaehne