hyperspy/rosettasciio #360: Problem loading dm3 files

j-m-ros, 2025-01-20:
#### Describe the bug
KeyError when loading Gatan dm3 files. 
Described as : BoxKeyError 'Meta Data'   in .../python3.11/site-packages/box/box.py#line=591 

#### To Reproduce
Error occurs when loading any Gatan dm3 file. All files tested have previously been readable. 
EMD files can be opened successfully.

Minimal code snippet: 
```
import hyperspy.api as hs
s=hs.load("Test/BFTEM_Y2O3.ZrO2_600000X_0010.dm3")
```

#### Expected behavior
File should  load as hyperspy signal

#### Python environement:
Tested with:
 - RosettaSciIO version: 0.6
 - Python version: 3.11.9
 - HyperSpy version: 2.1.1
also occurs with:
 - RosettaSciIO version: 0.7.1
 - Python version: 3.12.8
 - HyperSpy version: 2.2.0

ericpre, 2025-01-20:
It sounds like https://github.com/hyperspy/rosettasciio/issues/355.

Can you check what is your version of [python-box](https://pypi.org/project/python-box)? Version 7.3.1 has a bug, any other version higher than 6 works fine.

j-m-ros, 2025-01-20:
Bingo!
Thanks for the rapid response, yes, I had python-box 7.3.1 and reverting to 7.3.0 resolved the problem.
:-)

closed 2025-01-20 by ericpre