hyperspy/rosettasciio #364: Unable to load .dm3

colbaltapier, 2025-02-01:
#### Describe the bug
Unable to load .dm3 file. 

#### To Reproduce
Steps to reproduce the behavior:
```
import hyperspy.api as hs

s = hs.load('/home/harry/Documents/HL_Spectra.dm3')
````
#### Expected behavior
The expected behaviour is that the .dm3 file can be loaded into hyperspy. 

#### Python environement:
 - RosettaSciIO version: 0.7.1
 - Python version: 3.13.1
 - HyperSpy version: 2.2.0 (if relevant)
 - Exspy version: 0.3.1

#### Additional context
Add any other context about the problem here. If necessary, add screenshots to help explain your problem.
I checked the .dm3 file was not corrupt by opening it in ImageJ. 

The error message from the python console:
`ERROR | Hyperspy | If this file format is supported, please report this error to the RosettaSciIO developers at https://github.com/hyperspy/rosettasciio/issues (hyperspy.io:599)
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File ~/hyperspy/lib/python3.13/site-packages/spyder_kernels/customize/utils.py:209, in exec_encapsulate_locals(code_ast, globals, locals, exec_fun, filename)
    207     if filename is None:
    208         filename = "<stdin>"
--> 209     exec_fun(compile(code_ast, filename, "exec"), globals, None)
    210 finally:
    211     if use_locals_hack:
    212         # Cleanup code

File ~/untitled0.py:11
      3 """
      4 Created on Fri Jan 31 16:45:02 2025
      5 
      6 @author: harry
      7 """
      9 import hyperspy.api as hs
---> 11 s = hs.load('/home/harry/Documents/HL_Spectra.dm3')

File ~/hyperspy/lib/python3.13/site-packages/hyperspy/io.py:537, in load(filenames, signal_type, stack, stack_axis, new_axis_name, lazy, convert_units, escape_square_brackets, stack_metadata, load_original_metadata, show_progressbar, **kwds)
    533         objects.append(signal)
    534 else:
    535     # No stack, so simply we load all signals in all files separately
    536     objects = [
--> 537         load_single_file(filename, lazy=lazy, **kwds) for filename in filenames
    538     ]
    540 if len(objects) == 1:
    541     objects = objects[0]

File ~/hyperspy/lib/python3.13/site-packages/hyperspy/io.py:596, in load_single_file(filename, **kwds)
    590     raise ValueError(
    591         "`reader` should be one of None, str, " "or a custom file reader object"
    592     )
    594 try:
    595     # Try and load the file
--> 596     return load_with_reader(filename=filename, reader=reader, **kwds)
    598 except BaseException:
    599     _logger.error(
    600         "If this file format is supported, please "
    601         "report this error to the RosettaSciIO developers at "
    602         "https://github.com/hyperspy/rosettasciio/issues"
    603     )

File ~/hyperspy/lib/python3.13/site-packages/hyperspy/io.py:632, in load_with_reader(filename, reader, signal_type, convert_units, load_original_metadata, **kwds)
    630 if signal_type is not None:
    631     signal_dict["metadata"]["Signal"]["signal_type"] = signal_type
--> 632 signal = dict2signal(signal_dict, lazy=lazy)
    633 signal = _add_file_load_save_metadata("load", signal, reader)
    634 path = _parse_path(filename)

File ~/hyperspy/lib/python3.13/site-packages/hyperspy/io.py:835, in dict2signal(signal_dict, lazy)
    832 elif signal_dimension == -1:
    833     # If not defined, all dimension are categorised as signal
    834     signal_dimension = signal_dict["data"].ndim
--> 835 signal = assign_signal_subclass(
    836     signal_dimension=signal_dimension,
    837     signal_type=signal_type,
    838     dtype=signal_dict["data"].dtype,
    839     lazy=lazy,
    840 )(**signal_dict)
    841 if signal._lazy:
    842     signal._make_lazy()

File ~/hyperspy/lib/python3.13/site-packages/hyperspy/_signals/signal2d.py:347, in Signal2D.__init__(self, *args, **kwargs)
    345 if kwargs.get("ragged", False):
    346     raise ValueError("Signal2D can't be ragged.")
--> 347 super().__init__(*args, **kwargs)

File ~/hyperspy/lib/python3.13/site-packages/hyperspy/signal.py:2481, in BaseSignal.__init__(self, data, **kwds)
   2479     self.isig = SpecialSlicersSignal(self, False)
   2480     self.events = Events()
-> 2481     self.events.data_changed = Event(
   2482         """
   2483         Event that triggers when the data has changed
   2484 
   2485         The event trigger when the data is ready for consumption by any
   2486         process that depend on it as input. Plotted signals automatically
   2487         connect this Event to its `BaseSignal.plot()`.
   2488 
   2489         Note: The event only fires at certain specific times, not everytime
   2490         that the `BaseSignal.data` array changes values.
   2491 
   2492         Parameters
   2493         ----------
   2494             obj: The signal that owns the data.
   2495         """,
   2496         arguments=["obj"],
   2497     )
   2498     self._load_dictionary(kwds)
   2500 if self._signal_dimension >= 0:
   2501     # We don't explicitly set the signal_dimension of ragged because
   2502     # we can't predict it in advance

File ~/hyperspy/lib/python3.13/site-packages/hyperspy/events.py:186, in Event.__init__(self, doc, arguments)
    183 self._suppressed_callbacks = set()
    185 if arguments:
--> 186     self._trigger_maker(arguments)

File ~/hyperspy/lib/python3.13/site-packages/hyperspy/events.py:228, in Event._trigger_maker(self, arguments)
    226 gl.update({"f": orig_f})  # Make sure it keeps the original!
    227 exec(wrap_code, gl, locals())
--> 228 new_f = locals()["trigger"]
    229 # Replace the trigger function with the new one
    230 if defaults:

KeyError: 'trigger'`

ericpre, 2025-02-01:
This issue is fixed in https://github.com/hyperspy/hyperspy/pull/3468. They will be a release in the coming years adding support for python 3.13.

jlaehne, 2025-02-01:
> This issue is fixed in [hyperspy/hyperspy#3468](https://github.com/hyperspy/hyperspy/pull/3468). They will be a release in the coming years adding support for python 3.13.

weeks not years :-)

colbaltapier, 2025-02-01:
Hi, 

Thanks both! I will role back my python install in the meantime. 

Thanks, 
Harry

closed 2025-02-01 by colbaltapier