hyperspy/rosettasciio #367: Can't install dev version on python3.12

jlaehne, 2025-02-06:
When upgrading my development `conda` environment to `python3.12` or `python3.13`, I can't install `rosettasciio` any more with `pip install -e .` and it seems to be caused by the `cython` parsing of the `bruker` plugin. Here is the relevant part of the error message:

```
...
      rsciio/bruker/unbcf_fast.c: In function '__pyx_pf_6rsciio_6bruker_10unbcf_fast_parse_to_numpy':
      rsciio/bruker/unbcf_fast.c:7318:3: warning: 'ma_version_tag' is deprecated [-Wdeprecated-declarations]
       7318 |   __Pyx_GetModuleGlobalName(__pyx_t_2, __pyx_n_s_np); if (unlikely(!__pyx_t_2)) __PYX_ERR(1, 319, __pyx_L1_error)
            |   ^~~~~~~~~~~~~~~~~~~~~~~~~
      In file included from /home/me/.conda/envs/hyperspy-dev/include/python3.12/dictobject.h:90,
                       from /home/me/.conda/envs/hyperspy-dev/include/python3.12/Python.h:64,
                       from rsciio/bruker/unbcf_fast.c:19:
      /home/me/.conda/envs/hyperspy-dev/include/python3.12/cpython/dictobject.h:22:34: note: declared here
         22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
            |                                  ^~~~~~~~~~~~~~
      rsciio/bruker/unbcf_fast.c:7367:3: warning: 'ma_version_tag' is deprecated [-Wdeprecated-declarations]
       7367 |   __Pyx_GetModuleGlobalName(__pyx_t_1, __pyx_n_s_np); if (unlikely(!__pyx_t_1)) __PYX_ERR(1, 321, __pyx_L1_error)
            |   ^~~~~~~~~~~~~~~~~~~~~~~~~
      /home/me/.conda/envs/hyperspy-dev/include/python3.12/cpython/dictobject.h:22:34: note: declared here
         22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
            |                                  ^~~~~~~~~~~~~~
      rsciio/bruker/unbcf_fast.c:7423:3: warning: 'ma_version_tag' is deprecated [-Wdeprecated-declarations]
       7423 |   __Pyx_GetModuleGlobalName(__pyx_t_1, __pyx_n_s_np); if (unlikely(!__pyx_t_1)) __PYX_ERR(1, 324, __pyx_L1_error)
            |   ^~~~~~~~~~~~~~~~~~~~~~~~~
      /home/me/.conda/envs/hyperspy-dev/include/python3.12/cpython/dictobject.h:22:34: note: declared here
         22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
            |                                  ^~~~~~~~~~~~~~
      rsciio/bruker/unbcf_fast.c:7479:3: warning: 'ma_version_tag' is deprecated [-Wdeprecated-declarations]
       7479 |   __Pyx_GetModuleGlobalName(__pyx_t_1, __pyx_n_s_np); if (unlikely(!__pyx_t_1)) __PYX_ERR(1, 327, __pyx_L1_error)
            |   ^~~~~~~~~~~~~~~~~~~~~~~~~
      /home/me/.conda/envs/hyperspy-dev/include/python3.12/cpython/dictobject.h:22:34: note: declared here
         22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
            |                                  ^~~~~~~~~~~~~~
      rsciio/bruker/unbcf_fast.c: In function '__pyx_pf___pyx_MemviewEnum___reduce_cython__':
      rsciio/bruker/unbcf_fast.c:9726:5: warning: 'ma_version_tag' is deprecated [-Wdeprecated-declarations]
       9726 |     __Pyx_GetModuleGlobalName(__pyx_t_4, __pyx_n_s_pyx_unpickle_Enum); if (unlikely(!__pyx_t_4)) __PYX_ERR(0, 13, __pyx_L1_error)
            |     ^~~~~~~~~~~~~~~~~~~~~~~~~
      /home/me/.conda/envs/hyperspy-dev/include/python3.12/cpython/dictobject.h:22:34: note: declared here
         22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
            |                                  ^~~~~~~~~~~~~~
      rsciio/bruker/unbcf_fast.c:9772:5: warning: 'ma_version_tag' is deprecated [-Wdeprecated-declarations]
       9772 |     __Pyx_GetModuleGlobalName(__pyx_t_5, __pyx_n_s_pyx_unpickle_Enum); if (unlikely(!__pyx_t_5)) __PYX_ERR(0, 15, __pyx_L1_error)
            |     ^~~~~~~~~~~~~~~~~~~~~~~~~
      /home/me/.conda/envs/hyperspy-dev/include/python3.12/cpython/dictobject.h:22:34: note: declared here
         22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
            |                                  ^~~~~~~~~~~~~~
      rsciio/bruker/unbcf_fast.c: In function '__pyx_memview_slice':
      rsciio/bruker/unbcf_fast.c:14909:3: warning: 'Py_OptimizeFlag' is deprecated [-Wdeprecated-declarations]
      14909 |   if (unlikely(!Py_OptimizeFlag)) {
            |   ^~
      In file included from /home/me/.conda/envs/hyperspy-dev/include/python3.12/Python.h:48:
      /home/me/.conda/envs/hyperspy-dev/include/python3.12/cpython/pydebug.h:13:37: note: declared here
         13 | Py_DEPRECATED(3.12) PyAPI_DATA(int) Py_OptimizeFlag;
            |                                     ^~~~~~~~~~~~~~~
      rsciio/bruker/unbcf_fast.c: In function '__pyx_pymod_exec_unbcf_fast':
      rsciio/bruker/unbcf_fast.c:22375:3: warning: 'ma_version_tag' is deprecated [-Wdeprecated-declarations]
      22375 |   __Pyx_GetModuleGlobalName(__pyx_t_1, __pyx_n_s_sys); if (unlikely(!__pyx_t_1)) __PYX_ERR(1, 6, __pyx_L1_error)
            |   ^~~~~~~~~~~~~~~~~~~~~~~~~
      /home/me/.conda/envs/hyperspy-dev/include/python3.12/cpython/dictobject.h:22:34: note: declared here
         22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
            |                                  ^~~~~~~~~~~~~~
      rsciio/bruker/unbcf_fast.c: In function '__Pyx_Raise':
      rsciio/bruker/unbcf_fast.c:23089:34: error: 'PyThreadState' {aka 'struct _ts'} has no member named 'curexc_traceback'
      23089 |         PyObject* tmp_tb = tstate->curexc_traceback;
            |                                  ^~
      rsciio/bruker/unbcf_fast.c:23092:19: error: 'PyThreadState' {aka 'struct _ts'} has no member named 'curexc_traceback'
      23092 |             tstate->curexc_traceback = tb;
            |                   ^~
      rsciio/bruker/unbcf_fast.c: In function '__Pyx_get_tp_dict_version':
      rsciio/bruker/unbcf_fast.c:23575:5: warning: 'ma_version_tag' is deprecated [-Wdeprecated-declarations]
      23575 |     return likely(dict) ? __PYX_GET_DICT_VERSION(dict) : 0;
[... 88 lines cut ...]
      rsciio/bruker/unbcf_fast.c:27030:53: error: 'PyLongObject' {aka 'struct _longobject'} has no member named 'ob_digit'
      27030 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
            |                                                     ^~
      rsciio/bruker/unbcf_fast.c: In function '__Pyx_PyInt_As_char':
      rsciio/bruker/unbcf_fast.c:27209:53: error: 'PyLongObject' {aka 'struct _longobject'} has no member named 'ob_digit'
      27209 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
            |                                                     ^~
      rsciio/bruker/unbcf_fast.c:27264:53: error: 'PyLongObject' {aka 'struct _longobject'} has no member named 'ob_digit'
      27264 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
            |                                                     ^~
      rsciio/bruker/unbcf_fast.c: In function '__Pyx_PyIndex_AsSsize_t':
      rsciio/bruker/unbcf_fast.c:27615:45: error: 'PyLongObject' {aka 'struct _longobject'} has no member named 'ob_digit'
      27615 |     const digit* digits = ((PyLongObject*)b)->ob_digit;
            |                                             ^~
      Traceback (most recent call last):
        File "/tmp/pip-build-env-qdqyn5m7/overlay/lib/python3.12/site-packages/setuptools/_distutils/spawn.py", line 70, in spawn
          subprocess.check_call(cmd, env=_inject_macos_ver(env))
        File "/home/me/.conda/envs/hyperspy-dev/lib/python3.12/subprocess.py", line 413, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['/home/me/.conda/envs/hyperspy-dev/bin/gcc', '-pthread', '-B', '/home/me/.conda/envs/hyperspy-dev/compiler_compat', '-fno-strict-overflow', '-DNDEBUG', '-O2', '-Wall', '-fPIC', '-O2', '-isystem', '/home/me/.conda/envs/hyperspy-dev/include', '-fPIC', '-O2', '-isystem', '/home/me/.conda/envs/hyperspy-dev/include', '-fPIC', '-I/home/me/.conda/envs/hyperspy-dev/include/python3.12', '-c', 'rsciio/bruker/unbcf_fast.c', '-o', '/tmp/tmpmotbr_54.build-temp/rsciio/bruker/unbcf_fast.o']' returned non-zero exit status 1.
....
      error: command '/home/me/.conda/envs/hyperspy-dev/bin/gcc' failed with exit code 1

...
```

ericpre, 2025-02-07:
This is surprising, because python 3.12 has been used for a while now (on CI, packaging, user install or dev install). python 3.13 is more recent but it seems to work fine on CI and when testing locally.

They must be something in your local install. Maybe try to figure out if your are using a C compiler or some dependencies that are not supported? Use CI as baseline to know what works?
Alternatively, it could be a "corrupted" install, something in setting path, activating python environment.

ericpre, 2025-03-01:
@jlaehne, anything to follow up on this issue?

jlaehne, 2025-03-01:
Well, I have reproduced the issue on a separate machines - though with rather similar setup - and in a fresh conda environment.

Generally, it works as long as I use the released packages from conda (tested with python 3.12) or from the arch repositories (even with python 3.13). Just in my dev environment where I install from a local fork from github, I'm having trouble.

Closing as I have no time to dig deeper at the moment and the CIs all work fine.

closed 2025-03-01 by jlaehne

ericpre, 2025-03-02:
Could it be related to https://github.com/cython/cython/issues/5238? Which version of cython are you using? On python 3.12, it seems that cython>=3 is required.