hyperspy/rosettasciio #382: Error reading Bruker .bcf file

aloboa, 2025-03-22:
This is my 1st attempt after having installed hyperspy with:
$ conda create -n hyperspy
$ conda activate hyperspy
$ conda install hyperspy -c conda-forge
(hyperspy) alobo@pop-os:~/hyppy3$ conda install -c conda-forge hyperspyui

The I start python and try to follow
https://hyperspy-temp.readthedocs.io/en/latest/user_guide/io.html#bruker-composite-file

```
>>> import hyperspy.api as hs
>>> hs.load("/media/alobo/SP PHD U3/Islandia/Alteration/MappingsSEM/JoaquinSEM/DI17_mapping.bcf")
ERROR | Hyperspy | If this file format is supported, please report this error to the RosettaSciIO developers at https://github.com/hyperspy/rosettasciio/issues (hyperspy.io:600)
Traceback (most recent call last):
  File "<python-input-3>", line 1, in <module>
	hs.load("/media/alobo/SP PHD U3/Islandia/Alteration/MappingsSEM/JoaquinSEM/DI17_mapping.bcf")
	~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alobo/miniconda3/envs/hyperspy/lib/python3.13/site-packages/hyperspy/io.py", line 538, in load
	load_single_file(filename, lazy=lazy, **kwds) for filename in filenames
	~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alobo/miniconda3/envs/hyperspy/lib/python3.13/site-packages/hyperspy/io.py", line 597, in load_single_file
	return load_with_reader(filename=filename, reader=reader, **kwds)
  File "/home/alobo/miniconda3/envs/hyperspy/lib/python3.13/site-packages/hyperspy/io.py", line 633, in load_with_reader
	signal = dict2signal(signal_dict, lazy=lazy)
  File "/home/alobo/miniconda3/envs/hyperspy/lib/python3.13/site-packages/hyperspy/io.py", line 862, in dict2signal
	signal.metadata.Markers[key] = markers_dict_to_markers(mp["Markers"][key])
								   ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
  File "/home/alobo/miniconda3/envs/hyperspy/lib/python3.13/site-packages/hyperspy/drawing/markers.py", line 1034, in markers_dict_to_markers
	fill = kwargs.pop("fill")
KeyError: 'fill'
```

sem-geologist, 2025-03-22:
> (hyperspy) ***@***.***:~/hyppy3$ conda install -c conda-forge hyperspyui
> Then I start python and try to follow

Interesting. Could you send the file itself?

aloboa, 2025-03-22:
Sure, thanks for your interest:
https://www.dropbox.com/scl/fi/q2k4hdny9eh18f7101kl3/DI17_mapping.bcf?rlkey=5vez1uz5szdw1pnsh1l5y50wl&dl=0

ericpre, 2025-03-23:
This is the same issue as https://github.com/hyperspy/hyperspy/pull/3501.

https://github.com/hyperspy/hyperspy/pull/3501 fixed the issue by setting the default when converting markers from the old (hyperspy v1) to the new marker API (hyperspy v2).

It may be good to update the bruker reader to use the new marker API.

aloboa, 2025-03-23:
> It may be good to update the bruker reader to use the new marker API.
Excuse my ignorance, I just installed Hyperspy a few days ago using conda (as described above). 
How can we update the bruker reader?

astewart-uk, 2025-03-23:
I am also having the same problem. 

Changing fill = kwargs.pop("fill") to fill = kwargs.pop("fill", None) in markers.py appears to solve this.

aloboa, 2025-03-23:
I have many `markers.py` in my hyperspy environment. Would you happen to know which one needs to be modified?

```
(base) alobo@pop-os:~/miniconda3/envs/hyperspy$ find . | grep markers.py
./lib/python3.13/site-packages/pip/_vendor/distlib/markers.py
./lib/python3.13/site-packages/pip/_vendor/packaging/markers.py
./lib/python3.13/site-packages/setuptools/_vendor/wheel/vendored/packaging/markers.py
./lib/python3.13/site-packages/setuptools/_vendor/packaging/markers.py
./lib/python3.13/site-packages/matplotlib/testing/_markers.py
./lib/python3.13/site-packages/matplotlib/markers.py
./lib/python3.13/site-packages/matplotlib/markers.pyi
./lib/python3.13/site-packages/pkg_resources/tests/test_markers.py
./lib/python3.13/site-packages/hyperspy/utils/markers.py
./lib/python3.13/site-packages/hyperspy/tests/drawing/test_markers.py
./lib/python3.13/site-packages/hyperspy/docstrings/markers.py
./lib/python3.13/site-packages/hyperspy/drawing/markers.py
./lib/python3.13/site-packages/packaging/markers.py
./lib/python3.13/site-packages/exspy/tests/drawing/test_plot_spectra_markers.py
```

ericpre, 2025-03-24:
> It may be good to update the bruker reader to use the new marker API.

Sorry for the confusion, my message was unclear, this was a message for us, the developer! 😅 The issue is known and it is now known how it should be fixed.

@aloboa, In your case, it may be best to wait that this is fixed or install a development version until this is released, if this can't wait. Editing libraries in the distribution is not recommended.

aloboa, 2025-03-24:
ok, thanks. Please post news here so that I can follow.  Thanks for your efforts. Let me know if there is anything that I could do.

aloboa, 2025-03-27:
> or install a development version until this is released,

How? Using this github or the main one (https://github.com/hyperspy/hyperspy)?

CSSFrancis, 2025-03-27:
'pip install git+https //github.com/hyperspy/rosettasciio' should work if you have git installed

aloboa, 2025-03-27:
ok, I'll wait.

aloboa, 2025-03-27:
Just out of curiosity, were you able to open the file I sent?
If so, is it possible to extract the mappings?

CSSFrancis, 2025-03-28:
> Just out of curiosity, were you able to open the file I sent?
> If so, is it possible to extract the mappings?
> 

You should be able to upgrade to the newest rosettasciio and this should work now.

ericpre, 2025-03-28:
Fixed in https://github.com/hyperspy/rosettasciio/pull/383.

aloboa, 2025-03-28:
After `$ conda activate /home/alobo/hyperspy-bundle/`, (My Hyperspy installation was done using
`bash HyperSpy-bundle-2025.03.11-Linux-x86\_64-Intel.sh`), I have tried
`(/home/alobo/hyperspy-bundle) alobo@pop-os:~/hyperspy-bundle/lib/python3.12/site-packages$ pip install git+https://github.com/hyperspy/rosettasciio`

and get:

```
Installing collected packages: rosettasciio
  Attempting uninstall: rosettasciio
    Found existing installation: rosettasciio 0.7.1
    Uninstalling rosettasciio-0.7.1:
      Successfully uninstalled rosettasciio-0.7.1
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pyxem 0.20.0 requires dask<=2024.8.0, but you have dask 2025.2.0 which is incompatible.
Successfully installed rosettasciio-0.8.1.dev1+gc238f02c

```

Then it works:
```
$ python
Python 3.12.9 | packaged by conda-forge | (main, Mar  4 2025, 22:48:41) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hyperspy.api as hs
>>> hs.load("/media/alobo/SP PHD U31/Islandia/Alteration/MappingsSEM/JoaquinSEM/DI17_mapping.bcf")
[<Signal2D, title: Ch 0, dimensions: (|1600, 1200)>, <Signal2D, title: Ch 0, dimensions: (|1600, 1200)>, <EDSSEMSpectrum, title: EDX, dimensions: (1600, 1200|2048)>]
```

and also does in the HyperSpyUI.

Many thanks.
Is it possible to install hyperspy with this version of rosettasciio on windows?

ericpre, 2025-03-29:
@aloboa, this is included in the 0.8 release, you can install the latest release as normal.