hyperspy/rosettasciio #391: arina reader

smribet, 2025-04-16:
#### Describe the functionality you would like to see.
The RosettaSciIO package is very helpful for importing data! Would be possible to add a reader for the Dectris Arina camera data? 

#### Additional information
We added a reader for the arina data in py4DSTEM.  The formatting is a little different, but this could be a helpful starting point. 
https://github.com/py4dstem/py4DSTEM/blob/dev/py4DSTEM/io/filereaders/read_arina.py

ericpre, 2025-04-17:
Sure, it would be possible! Would you be interested in adding it? 😃

CSSFrancis, 2025-04-17:
@smribet One thing that might also be useful is reaching out to Dectris and seeing if they would be interested in supporting the reader.  In my experience that is fairly helpful as they are going to be much more helpful when/if the file format changes and evolves.  It also means more coming from a user rather than generically coming from someone who works with Hyperspy/Rosettasciio.

smribet, 2025-04-24:
These are both good suggestions, thanks! I opened a PR for a basic arina reader, but I will see if Dectris would be interested in supporting this in the future. I tried to follow your style guide as best as I could (including with a little help from some AI).  I'd appreciate any suggestions and feedback on the PR.

CSSFrancis, 2025-04-30:
Keeping this open until #398 is merged

ericpre, 2025-05-28:
#398 has been merged but I would suggest to keep this issue open to keep track of other features that would be good to look at
- [ ] add lazy support - from https://github.com/hyperspy/rosettasciio/pull/398#pullrequestreview-2857212942
- [ ] check the default chunk size in relation to reading speed, this may be not straightforward to improve the reading speed - from https://github.com/hyperspy/rosettasciio/pull/398#issuecomment-2897840174