hyperspy/rosettasciio #409: asw file does not upload to the HyperSpyUI

tkozlecki, 2025-06-06:
#### Describe the bug
I am trying to open the JEOL asw file in HuperSpyUI, but it does not work. 

#### To Reproduce

Traceback (most recent call last):
  File "C:\Programy\HyperSpy-bundle\Lib\site-packages\hyperspyui\mainwindowhyperspy.py", line 552, in load
    sig = hs.load(escaped)
          ^^^^^^^^^^^^^^^^
  File "C:\Programy\HyperSpy-bundle\Lib\site-packages\hyperspy\io.py", line 537, in load
    load_single_file(filename, lazy=lazy, **kwds) for filename in filenames
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Programy\HyperSpy-bundle\Lib\site-packages\hyperspy\io.py", line 596, in load_single_file
    return load_with_reader(filename=filename, reader=reader, **kwds)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Programy\HyperSpy-bundle\Lib\site-packages\hyperspy\io.py", line 618, in load_with_reader
    file_data_list = importlib.import_module(reader["api"]).file_reader(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Programy\HyperSpy-bundle\Lib\site-packages\rsciio\jeol\_api.py", line 133, in file_reader
    return extension_to_reader_mapping[file_ext](filename, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Programy\HyperSpy-bundle\Lib\site-packages\rsciio\jeol\_api.py", line 159, in _read_asw
    path = node["ViewData"][k]["Filename"].split("\\")
           ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'Filename'
```

#### Python environement:
 - RosettaSciIO version: 1.x.x
 - Python version:  Python 3.12.8 | packaged by conda-forge
 - HyperSpyUI version:  2.0.2
 - HyperSpy-bundle-2025.01.19-Windows-x86_64-Intel

ericpre, 2025-06-06:
I can reproduce the issue and I have no idea. With working around this error, it goes a bit further, but there are other errors. I am wondering if this is a type of asw files that is not supported.
Does it happen for a specific subset of files? or for all of them acquired on a given instrument?

@nem1234, any idea?

tkozlecki, 2025-06-06:
Yes, all data obtained has the same issue. This is a JEOL microscope with an older Oxford Instruments EDS.