hyperspy/rosettasciio #429: 0.10.0 Release Tracker

jlaehne, 2025-07-25:
Do we want to make a 0.10 release? I am particularly interested in #328 being available in a released version to facilitate colleagues the use of this file reader. Are there any currently open PRs/issues that should definitely be addressed still?

ericpre, 2025-07-25:
Sounds good! I think that we need to sort out https://github.com/hyperspy/rosettasciio/pull/425 and https://github.com/hyperspy/hyperspy/pull/3528.

ericpre, 2025-07-26:
I pull out the regression fix from https://github.com/hyperspy/hyperspy/pull/3528 in https://github.com/hyperspy/hyperspy/pull/3533 to fix the failure in the rosettaciio test suite with the development branch of hyperspy.

ericpre, 2025-07-26:
I would suggest to merge #426 because it is ready to go and then it is ready to go?

@jlaehne, are you happy to do the release?

jlaehne, 2025-07-27:
See #431