hyperspy/rosettasciio #438: Create a reader/writer

CSSFrancis, 2025-08-12:
#### Describe the functionality you would like to see.

It would be nice to have a reader/writer such that you could:

```python

data, axes,metadata, original_metadata =  rsciio.load("file.ext", lazy=True)

rsciio.save("file.ext", data=data, axes=axes, metadata=metadata, original_metadata= original_metadata)
```

and automatically determine the extension like hyperspy does.

@ericpre Is there any reason against moving the code from hyperspy downstream for this?

@smribet This should be helpful for reading/writing to different formats without too much additional overhead as long as the axes are defined correctly.

ericpre, 2025-08-13:
It may be worth checking if this has been discussed during the split - I can't remember. One reason could be that the code in HyperSpy use a heuristic to select the reader that evolves over the years and that may not be suitable for rosettasciio.

ericpre, 2025-08-14:
Yes, sounds good. The only hurdle is to define the logic that define automatically which io plugin to case the file extension is used by several plugin. If this needs to specify in the load or save function, then the API may end up being a bit messy.

jlaehne, 2025-08-20:
If it exists that logic is indeed "grown over time" and possibly not too systematic. However, in many cases even in HyperSpy there is no automatic decision and the `reader`/`extension` keyword needs to be provided. However, for non-binary formats like hdf5 or xml it should be indeed be possible to automatically select the correct reader -- and it would make sense to have that functionality directly in RosettaSciIO.

hakonanes, 2025-08-20:
Having worked a lot with IO, I've found allowing readers to take a peek at the file useful. It works like so:
1. Multiple readers support the file extension, and no `reader: str` parameter with a unique reader name is given
2. Each potential reader can supply a quick function with the signature `can_read_file(path: str) -> bool`. Function restrictions:
    i. Have to open and close the file safely (try/except and/or context manager)
    ii. No processing, just check whether something only this reader expects in the file (say, a "manufacturer" HDF5 dataset with the value "kikuchipy")
3. If more than one reader can read the file, raise an error and require the reader name

This look-ahead function should solve some ambiguities. Readers can be updated on a need-to-have basis, as in it's not required for readers supporting a unique extension.